Ultimaker/Cura #511: Commandeering USB tty at startup

Ghostkeeper, 2015-10-28:
Johan reported this to me. He had a different connection open to an Ultimaker Original+ printer, with a different application. When Cura starts, the other program loses connection.

Platform: OSX
Version: 15.09.91

Steps to reproduce:

* Obtain an UMO+. Connect it to a computer via USB.
* Launch Pronterface (or something similar) and set up a connection to your printer.
* Launch Cura.

Expected results:
* Nothing happens in Pronterface. It retains connection.
* Cura is not able to obtain a USB connection. If both programs would be able to obtain a connection, g-code could mix up and this should not happen.

Observed results:
* Pronterface loses connection.
* Cura connects to the printer.

Note that I haven't tested this myself, but this is just a re-formatting of what Johan told me. Also, Johan didn't specify which program was making a USB connection (and I can't find him right now to ask).

Xisle, 2015-10-28:
See issue #70

Ghostkeeper, 2015-10-29:
Ah right, duplicate then.

closed 2015-10-29 by Ghostkeeper