Ultimaker/Cura #512: Layer View: gcode view or area view?

BagelOrb, 2015-10-28:
We need to address whether the layer view is a viewer of the actual gcode, or a viewer of the feature areas (infill areas, support areas etc.)

Currently it is still not a gcode viewer; for example the combined infill lines are not shown as combined and the coasted parts are viewed as if they are printed as normal.

The old Cura used the area information as an early approximation so that you can show some output quite quickly before showing the actual outputted gcode.
Also the areas were used to show lower layers than the currently shown layer, because these areas are generally lower poly than the way the areas are actually filled.

fieldOfView, 2015-10-28:
> The old Cura used the area information as an early approximation so that 
> you can show some output quite quickly

If at all possible, it would be great if new-Cura could do something similar. I think it would help a lot to reduce the "perceived slowness" of auto-slicing.

awhiemstra, 2015-10-28:
I have some ideas in regards to streaming more of the slice data. However, it will most likely make layer view slower as it will need to do dynamic updates to the mesh.

> We need to address whether the layer view is a viewer of the actual gcode, or a viewer of 
> the feature areas (infill areas, support areas etc.)

I do not want to make the layer view dependant on GCode, as said before. GCode is an output artifact, in the same way that the meshes we generate from the engine data are an output artifact. We take the data from a step before that and then produce a form of output. The fact that it is GCode that is being output should have no bearing on that all information needs to be available before doing the transform to GCode. And assuming all information is available, we can just as easily send that to Cura and transform that into a renderable mesh.

BagelOrb, 2015-10-29:
You misundertood me; by 'gcode' I just meant the eventual state of information - I am not referring to the format in which it has been saved. It's like I said 'STL' instead of 'mesh'. I should have said something like 'instruction set', but that still doesn't seem right.

The question is what stage in the process from mesh to instructions the layer view should be visualizing.
The following stages are roughly seperable in the engine:
0- mesh
1- mesh outlines
2- feature areas and outlines *
3- straight consequent path segments *
4- path segments in different line widths; coasting paths etc. (the eventual instructions as represented in the gcode)

The starred stages above are currently used in the layer view.

I am a proponent of using stage 2 only, but perhaps you disagree and want stage 4 only.
As it was in the old Cura would be a combination of stage 2 and 4...

Ghostkeeper, 2015-10-30:
Stage 0 is also already implemented!

For layer view, I think stage 2 is sufficient and would be the best option if we really want to keep it output-independent. But I also think that visualising the output is the primary function of layer view. I think it would be best to implement stage 4 eventually, even if it is dependent on g-code as medium. In the ideal, it would be another view in the drop-down menu, "output view".

BagelOrb, 2015-10-30:
@Ghostkeeper you misunderstood me; by 'gcode' I just meant the eventual state of information: stage 4.
Gcode does not have to be the medium for getting that data to the Cura frontend. 
All stages are output independent.

All stages (except 0) can send information with the current interface of the command socket. The command socket is the only medium under consideration in this github issue.

The question is simple: of what stage should the layer view be a viewer?

BagelOrb, 2016-03-07:
Answer: stage 4

closed 2016-03-07 by BagelOrb