Ultimaker/Cura #516: global settings vs machine-local setting visibility

BagelOrb, 2015-10-28:
When I goto [ Settings ] > [ Preferences ] I expect to find all global settings and not settings specific to the current setup.

However, when I change the machine to a multi-extrusion machine then the settings under [ Setting Visibility ] are different.

If the settings are machine-specific, I would think they should be accessed via [Settings] > [Preferences] > [Setting Visibility] > [MACHINE] > .... actual list of settings with checkboxes

Is this a UX bug?

BagelOrb, 2017-07-21:
The settings visibility in the preferences should either be categorized under a machine selection dropbox or hold for all printers the same.

Ghostkeeper, 2017-07-27:
The settings visibility list is the same for every machine, right? What is different?

BagelOrb, 2017-07-28:
Errr yes. Mindfuck here!

BagelOrb, 2017-07-28:
This is not an issue. Sorry to have opened it in the first place.

closed 2017-07-28 by BagelOrb