Ultimaker/Cura #517: [Settings] vs. [Preferences]

BagelOrb, 2015-10-28:
Currently the only place where the Preferences window can be accessed is via the [Settings] menu bar item,
while the Preferences window again contains a tab for [Settings].

This is quite confusing.

Please put the the Preferences menu item under [edit] (and remove the Settings menu bar item altogether) or rename the Settings menu bar item to something different.

Ghostkeeper, 2015-10-29:
I'm voting for renaming the Settings menu bar item to Preferences, though I can't currently see the menu bar at all.

BagelOrb, 2015-10-29:
But then it would be 
```
[ Preferences    ]
[ Preferences... ]
```
which seems a bit ugly

mpbomil, 2015-11-05:
You could rename the "Settings" tab in preferences to "Print" or "Printer", since it contains settings for the printer

BagelOrb, 2015-11-06:
The preferences also cover general preferences, view preferences and plugin preferences...

fieldOfView, 2016-11-09:
In Cura 2.3 (and beyond) the chosen solution is Preferences -> Configure Cura... -> Settings

This can be closed

closed 2016-11-09 by nallath