Ultimaker/Cura #522: [15.09.91] Layer height 0.15mm crashes slice

DexCisco, 2015-10-29:
I can slice a model with layer height at .1 or .2, but .15 causes the slice to crash half way through.

nallath, 2015-10-30:
Could you post your logs?

Ghostkeeper, 2015-10-30:
We have a known bug similar to this in version .91 which was fixed, but that didn't work at a layer height of 0.2mm either. There is also a related bug that causes it to crash when the infill is 0%, when combined with some layer height settings, which I'll be working on now.

I haven't heard of anything that makes it crash at 0.15mm layer height but not 0.1mm or 0.2mm.

DexCisco, 2015-10-30:
Calling engine with: ['C:\\Program Files\\Cura 15.09\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '-j', 'C:\\Program Files\\Cura 15.09\\resources\\machines\\fdmprinter.json', '-vv']

Cura_SteamEngine version DEV
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
Loading polygon setting machine_disallowed_areas not implemented...
Unrecognized data type in JSON: machine_platform_offset has type array
Loading polygon setting machine_head_polygon not implemented...
Loading polygon setting machine_head_with_fans_polygon not implemented...
Warning: setting an unregistered setting machine_head_polygon
Warning: setting an unregistered setting machine_head_with_fans_polygon
Warning: setting an unregistered setting machine_disallowed_areas
Warning: setting an unregistered setting machine_use_extruder_offset_to_offset_coords
Unregistered setting mesh_position_z
Unregistered setting mesh_position_y
Unregistered setting mesh_position_x
Unregistered setting center_object
Unregistered setting extruder_nr
 -g -e0 -e0 -l "0"
Unregistered setting support_extruder_nr
Unregistered setting support_roof_extruder_nr
Progress: start accomplished in 0.000s
Starting slice...
Slicing model...
Layer count: 112
Progress: slice accomplished in 0.078s
Starting layerparts...
Unregistered setting multiple_mesh_overlap
Progress: layerparts accomplished in 0.031s
Starting inset...
Progress: inset accomplished in 0.343s
Starting support...
Progress: support accomplished in 0.000s
Starting skin...

DexCisco, 2015-10-30:
2015-10-30 12:57:30,707 - INFO - Loaded plugin ConsoleLogger
2015-10-30 12:57:30,963 - INFO - Loaded plugin CuraEngineBackend
2015-10-30 12:57:30,984 - INFO - Loaded plugin OBJReader
2015-10-30 12:57:31,005 - INFO - Loaded plugin OBJWriter
2015-10-30 12:57:31,031 - INFO - Loaded plugin STLReader
2015-10-30 12:57:31,061 - INFO - Loaded plugin STLWriter
2015-10-30 12:57:31,097 - INFO - Loaded plugin FileLogger
2015-10-30 12:57:31,130 - INFO - Loaded plugin GCodeWriter
2015-10-30 12:57:31,165 - INFO - Loaded plugin LayerView
2015-10-30 12:57:31,204 - INFO - Loaded plugin LocalFileOutputDevice
2015-10-30 12:57:31,269 - INFO - Loaded plugin PostProcessingPlugin
2015-10-30 12:57:31,325 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-10-30 12:57:31,434 - INFO - Loaded plugin SliceInfoPlugin
2015-10-30 12:57:31,487 - INFO - Loaded plugin CameraTool
2015-10-30 12:57:31,550 - INFO - Loaded plugin MirrorTool
2015-10-30 12:57:31,612 - INFO - Loaded plugin PerObjectSettingsTool
2015-10-30 12:57:31,685 - INFO - Loaded plugin RotateTool
2015-10-30 12:57:31,745 - INFO - Loaded plugin ScaleTool
2015-10-30 12:57:31,790 - INFO - Loaded plugin SelectionTool
2015-10-30 12:57:31,842 - INFO - Loaded plugin TranslateTool
2015-10-30 12:57:31,888 - INFO - Checking for new version of cura
2015-10-30 12:57:31,888 - INFO - Loaded plugin UpdateChecker
2015-10-30 12:57:31,977 - INFO - Loaded plugin USBPrinting
2015-10-30 12:57:32,057 - INFO - Loaded plugin MeshView
2015-10-30 12:57:32,135 - INFO - Loaded plugin WireframeView
2015-10-30 12:57:32,175 - INFO - Started engine process: C:\Program Files\Cura 15.09\CuraEngine.exe
2015-10-30 12:57:32,283 - DEBUG - Setting value of setting layer_height to 0.08
2015-10-30 12:57:32,284 - WARNING - Tried to set value of non-user setting layer_height
2015-10-30 12:57:32,287 - DEBUG - Setting value of setting layer_height to 0.15
2015-10-30 12:57:32,287 - WARNING - Tried to set value of non-user setting layer_height
2015-10-30 12:57:32,290 - DEBUG - Setting value of setting layer_height to 0.1
2015-10-30 12:57:32,291 - WARNING - Tried to set value of non-user setting layer_height
2015-10-30 12:57:32,294 - DEBUG - Setting value of setting layer_height to 0.06
2015-10-30 12:57:32,294 - WARNING - Tried to set value of non-user setting layer_height
2015-10-30 12:57:32,310 - DEBUG - Setting active view to MeshView
2015-10-30 12:57:32,371 - DEBUG - Backend connected on port 49674
2015-10-30 12:57:34,533 - DEBUG - Loaded theme C:\Program Files\Cura 15.09\resources\themes\cura
2015-10-30 12:57:37,694 - DEBUG - Preparing to send slice data to engine.
2015-10-30 13:03:50,758 - DEBUG - Calculating normals took 0.013004064559936523 seconds
2015-10-30 13:03:50,759 - DEBUG - Loaded a mesh with 47004 vertices
2015-10-30 13:03:50,845 - DEBUG - Loading mesh took 0.5949809551239014 seconds
2015-10-30 13:03:51,557 - DEBUG - Preparing to send slice data to engine.
2015-10-30 13:03:51,636 - DEBUG - Sending data to engine for slicing.
2015-10-30 13:04:06,181 - DEBUG - Preparing to send slice data to engine.
2015-10-30 13:04:12,861 - DEBUG - Calculating normals took 0.014000892639160156 seconds
2015-10-30 13:04:12,862 - DEBUG - Loaded a mesh with 51216 vertices
2015-10-30 13:04:12,940 - DEBUG - Loading mesh took 0.7040410041809082 seconds
2015-10-30 13:04:13,665 - DEBUG - Preparing to send slice data to engine.
2015-10-30 13:04:13,745 - DEBUG - Sending data to engine for slicing.
2015-10-30 13:04:19,273 - DEBUG - Setting value of setting layer_height to 0.15
2015-10-30 13:04:23,344 - DEBUG - Killing engine process
2015-10-30 13:04:23,904 - INFO - Backend crashed or closed. Restarting...
2015-10-30 13:04:24,174 - INFO - Started engine process: C:\Program Files\Cura 15.09\CuraEngine.exe
2015-10-30 13:04:24,235 - DEBUG - Backend connected on port 49674
2015-10-30 13:04:24,730 - DEBUG - Preparing to send slice data to engine.
2015-10-30 13:04:24,814 - DEBUG - Sending data to engine for slicing.
2015-10-30 13:04:41,747 - INFO - Backend crashed or closed. Restarting...
2015-10-30 13:04:42,009 - INFO - Started engine process: C:\Program Files\Cura 15.09\CuraEngine.exe
2015-10-30 13:04:42,064 - DEBUG - Backend connected on port 49674

BagelOrb, 2015-11-04:
Can you post the model file?

Which version of Cura are you using?

DexCisco, 2015-11-04:
The Dev version seems to work.  I was using 15.09.91

closed 2015-11-04 by DexCisco