Ultimaker/Cura #523: Infill Twice on Same LAyer

ChrisKrieg, 2015-10-30:
Just noticed this really weird gcode glitch that is making it quite difficult to print a part at 90% infill.

Cura prints the perimeters first, as it should.
Cura prints the infill at 90% fill, correctly.
Cura prints the solid fill for top/bottom solid sections (printing an arch), correctly.
Cura prints the 90% infill again. This causes the infill section to pillow upward and the nozzle crashes into the overflow after enough layers build up the fill.

Another really weird quirk is that it only double-fills the regions of the arch legs. it fills the keystone region of the arch, after the bridging operation, only once (correct).

I would attach the gcode if I could, but I don't have write permission.

BagelOrb, 2015-11-04:
Which version of Cura are you using?

BagelOrb, 2015-11-04:
The way you describe it it sounds like a bug that has been solved a couple of months ago.

closed 2015-11-04 by BagelOrb

ChrisKrieg, 2015-11-04:
I'm on mobile so it's hard to reply at the moment.

I'm using 15.04.03 and I narrowed down the issue. The problem arises if infill is less than 100%. When doing solid infill cura only makes one infill pass, but if infill is sparse it will finish the initial infill and top/bottom solid infill regions, then it will go back and infill the sparse regions an extra time.

I have only seen the problem when using infill of 90%. When I use 40% infill the issue does not occur.

It appeared that in some parts the problem arose between the overlap between sparse infill and top/bottom solid infill regions. The solid and sparse infill would overlap so far that it would cause an infill overflow issue. This only became noticeable at high infill because the excess had nowhere to go.

BagelOrb, 2015-11-04:
This repository is for the new Cura (version >= 15.06); you should post your issue on https://github.com/daid/Cura .

However, I've never heard anyone had that issue on that version. Are you sure you aren't using 15.06?