Ultimaker/Cura #524: [15.09.91] Only retraction settings show in Material section

DexCisco, 2015-10-30:
I have checked off all of the options in the material section of the settings visibility preferences, but nothing above enable retraction is showing.  e.g. Temps, diameter, flow.

DexCisco, 2015-10-30:
After disabling all the shown settings, the material section says "4 hidden settings use custom values".  I assume that this is the UM2 profile overriding the settings because using a different printer profile shows those settings.  So these settings are not adjustable for the UM2?

awhiemstra, 2015-11-02:
Material settings are handled by the printer on the UM2. Cura does not allow overriding the material settings when using an UM2. The "X hidden settings use customised values" message should not be shown though, so that part is a bug.

EldRick, 2015-11-02:
This also gets pretty screwed up when selecting a Prusa printer profile, with the Bed Temp option vanishing after first use.

closed 2016-09-17 by thopiekar