Ultimaker/Cura #525: [15.09.91] Super slowdown during print

DexCisco, 2015-10-30:
At various times during printing gcode from SD card, the printer head slows down to less than 1mm/s, both during printing and subsequent travels.  Eventually it wakes up and resumes at normal speeds.  The slowdown causes the model to melt under the nozzle and causes lots of stringing during travels, not to mention makes the prints take several times longer.

gcode link below:
https://drive.google.com/file/d/0B4z51yLR6zrQZlcyUi1NaGFwUk0/view?usp=sharing

nallath, 2015-11-02:
What machine are you using?

BagelOrb, 2015-11-04:
I haven't seen this bug in quite a while. It should have been fixed a while ago.

Which version of Cura are you using?

Can you maybe provide the stl file of this object?

DexCisco, 2015-11-04:
As above, I was using 15.09.91 on a UM2.  I noticed the slowdown on two different prints.  I will try my next print using the dev version.

BagelOrb, 2015-11-04:
Please provide me with the STL file.

If the STL is not open source, it would be nice if you could provide it to me via mail at t.kuipers(at)ultimaker.com

BagelOrb, 2015-11-04:
Did you use any settings other than the standard ones?

BagelOrb, 2015-11-06:
Thanks for sending me the STL.

In your mail you say you have been "playing with the retraction and travel settings".
Looking at your gcode it seems you also turned on coasting; is that right?

I think I'm mostly seeing unretraction moves at slow speeds - unretraction moves caused by coasting.

I will be trying to reproduce your problem, but it might help me a lot to know what kind of settings you have tried. What kind of non-standard retraction and travel settings did you try? Which settings did you tweak?

BagelOrb, 2015-11-06:
At line 6038 of your gcode it says:
```
G1 F60.000 E693.92022
G1 X119.837 Y115.312 E693.94131
```

The unretraction speed is set, but the new speed for the next line segment isn't set properly.

Ghostkeeper, 2015-11-09:
This was fixed in commits Ultimaker/CuraEngine@09afe2264a32356b4cd0083fd2a4f9cd3f6b5758 and Ultimaker/CuraEngine@087efefc1655bfd9f9a77caed6977ef2af4e039e. It's been tested and seems to work now.

closed 2015-11-09 by Ghostkeeper