Ultimaker/Cura #526: [15.09.91] just crash

robomaniac, 2015-10-31:
So, I just downloaded 15.09.91 and it crash.

[image]

I uninstall and reinstall same problem.

robomaniac, 2015-10-31:
Calling engine with: ['C:\\Program Files\\Cura 15.09\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '-j', 'C:\\Program Files\\Cura 15.09\\resources\\machines\\fdmprinter.json', '-vv']

Cura_SteamEngine version DEV
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.
Loading polygon setting machine_disallowed_areas not implemented...
Unrecognized data type in JSON: machine_platform_offset has type array
Loading polygon setting machine_head_polygon not implemented...
Loading polygon setting machine_head_with_fans_polygon not implemented...

robomaniac, 2015-10-31:
installed 15.09.90 and I can now print.

closed 2015-11-04 by BagelOrb

BagelOrb, 2015-11-05:
fixed for next release