Ultimaker/Cura #528: wrong Z heigth if model scaled

EugeneSkrebnev, 2015-11-01:
https://gyazo.com/172c947990ff412e7cec8d9c7d313b80

got model 1x1x1 cube scaled by 200%
got 37 layers with 0.3mm layer
37*0.3 = 11.1 not 20.0

XY scaled right.
this is because after scale model Z didnt change and part of detail moved under build plate

https://gyazo.com/c3794dcff59d9774415868c44da1de66

EugeneSkrebnev, 2015-11-01:
15.09.91

Ghostkeeper, 2015-11-02:
Bug in the platform collision code. It should move the model up, but I'm guessing this is an edge case since exactly half is below the platform.

You can pull it up on the platform with the upwards arrow when you deactivate the scale tool. It should drop the object down on the platform when you pull it too high.

EugeneSkrebnev, 2015-11-02:
yep. Once i realised that object is under plate i use reset positions to deal with it.

awhiemstra, 2015-11-02:
This should be resolved with the latest build, there were some issues with the code pushing objects back up.

closed 2015-11-02 by awhiemstra