Ultimaker/Cura #530: Please create an AppData file for Cura

hughsie, 2015-11-03:
Please consider writing and installing an AppData file with the application description and some screenshots, else Cura looks really bad in the GNOME and KDE Software Centers. We'd love to showcase more applications, but without the extra data file we can't. I use Cura quite a lot, and would love to feature it on the front page. If you need help or an example file, please just ask.

See http://people.freedesktop.org/~hughsient/appdata/ for details; thanks!

Richard

nallath, 2015-11-04:
Shouldn't take too much effort I guess.

hughsie, 2016-10-06:
I'm getting reports from people that cura isn't visible in gnome-software and I'm pointing them to this bug. Would you like me to try to create an AppData PR for this functionality?

nallath, 2016-10-06:
Please do. We're very close to a release, so our minds are focused on other things right now. 
Sorry that this issue fell through the cracks

closed 2016-10-10 by Ghostkeeper