Ultimaker/Cura #531: Run CuraEngine.exe in command prompt

sarahSamonte, 2015-11-03:
I want to use the backend: CuraEngine to try to compute the printing time of STL files programatically. However, when I try to open run it in the command prompt, I got this error message:

ERROR: Failed to load json file: fdmprinter.json
Unregistered setting machine_end_gcode
Unregistered setting speed_travel
maxObjectHeight : 0
M107
G0 X0 Y0 Z5

I do not have a 3d printer connected to my computer. I am using windows 8.1.

nallath, 2015-11-04:
Well, its fairly obvious what it requires from you right? It can't open the fdmprinter.json file. Are you sure that you used the right parameters to open it?

Ghostkeeper, 2015-11-04:
When you call `CuraEngine.exe`, you have to provide the argument `-j <JSON file>` that specifies where the JSON file is located.

closed 2015-11-04 by BagelOrb

sarahSamonte, 2015-11-12:
So now I was able to run it without the error in the .json file. But I got this:

[image]

I tried running Cura Engine 15.04 but it properly opened the stl file but 15.06 did not..

BagelOrb, 2015-11-12:
15.06 was a beta candidate (although this was not properly communicated to the community).
Try the latest release: 	Cura-1.99.0-dev.20151111-win64.exe .
You can find it at http://software.ultimaker.com/Cura_closed_beta/15.10/

This is the beta release of Cura 2.0, which will be launched very soon.