Ultimaker/Cura #532: Save to File button stops working after using it from the Layer View

ErikDeBruijn, 2015-11-05:
I'm able to save GCode from Cura. The Save dialog will show up (multiple times, if I repeat it). However, once I've switched to the layer view the button doesn't do anything anymore, also when I switch back to the regular view.

Both the button and save all menu feature don't give a response:
[image]

After this, no models will save anymore.

Strange thing, if I don't press "Save" while in the Layer View, it will work when I switch to the Mesh View again.

I can reproduce it every time. Reach out if I can help debug.

My Logs:

```
Nov  5 10:05:40  Cura[49643] <Notice>: 2015-11-05 10:05:40,109 - DEBUG - Setting value of setting layer_height to 0.1
Nov  5 10:05:40  Cura[49643] <Notice>: 2015-11-05 10:05:40,110 - DEBUG - Setting value of setting layer_height to 0.08
Nov  5 10:05:40  Cura[49643] <Notice>: 2015-11-05 10:05:40,110 - DEBUG - Setting value of setting layer_height to 0.15
Nov  5 10:05:40  Cura[49643] <Notice>: 2015-11-05 10:05:40,111 - DEBUG - Setting value of setting layer_height to 0.1
Nov  5 10:05:40  Cura[49643] <Notice>: 2015-11-05 10:05:40,112 - DEBUG - Setting value of setting layer_height to 0.06
Nov  5 10:05:40  Cura[49643] <Notice>: 2015-11-05 10:05:40,131 - DEBUG - Setting active view to MeshView
...
Nov  5 10:05:41  Cura[49643] <Notice>: 2015-11-05 10:05:41,121 - DEBUG - Loaded theme /Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/cura/resources/themes/cura
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/UM/Preferences/ProfilesPage.qml:59:23: Unable to assign [undefined] to int
Nov  5 10:05:43  Cura[49643] <Notice>: 2015-11-05 10:05:43,363 - INFO - Started engine process: /Applications/Cura-1.99.0-dev.20151101.app/Contents/MacOS/CuraEngine
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
Nov  5 10:05:43  Cura[49643] <Notice>: file:///Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/qml/QtQuick/Controls/Styles/Base/ButtonStyle.qml:154: TypeError: Cannot read property of null
Nov  5 10:05:43  Cura[49643] <Notice>: 2015-11-05 10:05:43,499 - DEBUG - Backend connected on port 49675
Nov  5 10:05:43  Cura[49643] <Notice>: 2015-11-05 10:05:43,517 - DEBUG - Preparing to send slice data to engine.
Nov  5 10:05:48  Cura[49643] <Notice>: QFileInfo::absolutePath: Constructed with empty filename
Nov  5 10:05:51  Cura[49643] <Notice>: 2015-11-05 10:05:51,055 - DEBUG - Calculating normals took 0.0003719329833984375 seconds
Nov  5 10:05:51  Cura[49643] <Notice>: 2015-11-05 10:05:51,056 - DEBUG - Loaded a mesh with 1548 vertices
Nov  5 10:05:51  Cura[49643] <Notice>: 2015-11-05 10:05:51,059 - DEBUG - Loading mesh took 0.11521291732788086 seconds
Nov  5 10:05:51  Cura[49643] <Notice>: 2015-11-05 10:05:51,747 - DEBUG - Preparing to send slice data to engine.
Nov  5 10:05:51  Cura[49643] <Notice>: 2015-11-05 10:05:51,764 - DEBUG - Sending data to engine for slicing.
Nov  5 10:06:02  Cura[49643] <Notice>: 2015-11-05 10:06:02,266 - DEBUG - Writing to Local File /Applications/Cura-1.99.0-dev.20151101.app/Contents/Resources/UM2_DoorHanger40.gcode in text mode
Nov  5 10:06:08  Cura[49643] <Notice>: 2015-11-05 10:06:08,034 - DEBUG - Preparing to send slice data to engine.
Nov  5 10:06:08  Cura[49643] <Notice>: 2015-11-05 10:06:08,055 - DEBUG - Sending data to engine for slicing.
Nov  5 10:06:18  Cura[49643] <Notice>: 2015-11-05 10:06:18,554 - DEBUG - Setting active view to LayerView
Nov  5 10:06:26  Cura[49643] <Notice>: 2015-11-05 10:06:26,369 - DEBUG - Setting active view to MeshView
Nov  5 10:07:57  Cura[49643] <Notice>: 2015-11-05 10:07:57,980 - DEBUG - Setting active view to LayerView
Nov  5 10:08:00  Cura[49643] <Notice>: 2015-11-05 10:08:00,625 - DEBUG - Setting active view to MeshView
```

Ghostkeeper, 2015-11-05:
Sounds similar to what @BagelOrb experienced.

mpbomil, 2015-11-05:
I can reproduce this bug also. When I switch to another menu item (like Save to Removable Drive) and then go back to Save to file, it works again.

fieldOfView, 2015-11-11:
Also possibly related:
* Press "Save to File" button
* Cancel the file dialog
* Press "Save to File" again

Result: File dialog does not reappear.
Workaround: select "Save to Removable drive", then select "Save to File" again.

BagelOrb, 2017-07-21:
Fixed a loooong time ago

closed 2017-07-21 by BagelOrb