Ultimaker/Cura #534: How to Build Cura from localhost source code?

shawnlintw, 2015-11-11:
Hello !

I want to build my self's Cura version.
I know the Cura provide a build script : cura_build.
However, I found the cura_build is download the source code on Github to compiler.
I want to modify the source to add new function therefore, I want to know how to build a special version from localhost source.(On my computer).

Sorry, I am a makefile beginner. 
Could you give me some instructions?

fieldOfView, 2015-11-11:
What platform is your computer? I "jerry-rigged" a setup that lets me run Cura from source, on windows.

You need to install Python 3, PyQt, and a couple more dependencies (ProtoBuf was especially tricky to find a binary version of). Then you need to check out the following repos:
Cura
Uranium
libArcus
I also added a "bin" folder containing the binaries for the CuraEngine executables and the dlls it needs. I picked those from a recent release. Finally you need to add some paths to the PythonPath environment variable.

Do you have any experience with Python?

shawnlintw, 2015-11-12:
Hello, 
Thanks "fieldOfView". I have one years experience with Python development (in Web App and Game)
My platform is PC Windows 7 64bits, mingw32-make with 32bit python3.4 and pyqt 32bits.

We want to develop the Cura for touch panel (win-PC), therefore we need build cura from source code.

Could anyone give me any suggest or Makefile example that I can build application from localhost source code ?

Thanks a lot.

Ghostkeeper, 2015-11-12:
It might be possible to let the cura-build script run its course, then modify the source code. If it's just the Python and QML, you possibly won't even need to compile anything again with cura-build's CMake script.

As far as I know, all of Ultimaker's developers work with the separate source repositories, as described by @fieldOfView. Each of the repositories has its own CMake list. With CMake, this list is translated into a Makefile which will install the program. So install CMake and open CMake-GUI, and point it to the source directory of each of the repositories (libArcus, [CuraEngine,] Uranium, Cura, in that order). Click configure, then generate, and then call mingw32-make on the Makefile. To get any further you'll need packaging software, which I'm sure @awhiemstra can help with.

You will also probably need MinGW-w64 to compile CuraEngine and libArcus from source.

iashwinprasada, 2015-11-30:
Hey,
even i was having a lot of trouble getting to build this for  windows system, but in the last two days ive got quite far, like @Ghostkeeper  suggested, i individually troubleshooted project by project...

but now im stuck at Qt

im getting 

fatal error: direct.h: No such file or directory
 #include <direct.h>
[image]

for debugging, when i comment that line and run make, im getting QT_OPEN, QT_CLOSE not found
any idea how to fix this?
[image]

Ghostkeeper, 2015-12-01:
Strange bug, that one! It's in Qt and I haven't seen it before.

I found [this](http://stackoverflow.com/questions/6193607/cannot-find-direct-h) link which says that you could try removing the include directive from the file, but that StackOverflow answer has no relation with Qt I don't think. Also, I've tried compiling with MinGW too and got no such error.

fieldOfView, 2015-12-01:
Since I am not all that interested in Qt issues, I "cheated" and use the binary package of PyQt provided here:
https://riverbankcomputing.com/software/pyqt/download5
This includes PyQt, Qt and lots of smaller dependencies of both of those.

iashwinprasada, 2015-12-01:
Yeah, in the screenshot I have attached I did run it by removing the directive...

Anyway I installed qt separately and removed it from cmake , now I'm getting other errors for sip...

I think it can be solved, I'll update what happened!

iashwinprasada, 2015-12-01:
hello! im stuck at a new point now, i have gotten a bit further. but im just writting the procedure i followed until now, to help others save some time

1. Download cura and cura-build into cura folder
    + https://github.com/Ultimaker/Cura
    + https://github.com/Ultimaker/cura-build

_________________
2. From that directory,
	  + run "env_win32.bat" when doing 32-bit builds
	  + run "env_win64.bat" for 64-bit builds.

_________________
3. Run cmake from inside cura-build-master
    + using MinGW generator
    + from GUI or < cmake -G "MinGW Makefiles" >

_________________
4. Call "mingw32-make" to build everything
  - Protobuf
    + was getting proto3 syntax error because proto2 was being built, had to find latest binary for proto3 n replace in Mingw/bin
    + c++ vcdist.dll not found error, install visual studio and move down python 64 to 32
    + cura-build-master\Protobuf-python-prefix\src\Protobuf-python\python\setup.py
      i had to comment ext_module_list.apped because i was getting this error

      cl : Command line error D8021 : invalid numeric argument '/Wno-write-strings'
    + somehow the env_win32.bat doesnt seem to effect, hence manually setting
    export PYTHONPATH="C:\cygwin64\local\Cura-master\cura-build-master\inst\lib\python3.4\site-packages"
  - Arcus

    - Could NOT find PROTOBUF (missing:  PROTOBUF_LIBRARY)
      + set in cmake PHOTOBUF_LIBRARY C:/cygwin64/local/Cura-master/cura-build-master/inst/lib

      + set PHOTOBUF_INCLUDE_DIR C:/cygwin64/local/Cura-master/cura-build-master/inst/include

### But Stuck here now
 - CuraEngine
      + g++: fatal error: no input files
compilation terminated.
CMakeFiles\_CuraEngine.dir\build.make:70: recipe for target 'CMakeFiles/_CuraEngine.dir/src/bridge.cpp.obj' failed
mingw32-make[5]: *** [CMakeFiles/_CuraEngine.dir/src/bridge.cpp.obj] Error 1
[image]

fieldOfView, 2015-12-02:
Again, I cheat... I use a compiled version of CuraEngine from a recent build. If you are only interested in making changes to the GUI (like I am), you don't necessarily have to compile the CuraEngine from source.

Obvious question: Did you check out the CuraEngine repository?

Ghostkeeper, 2015-12-02:
You certain you've set `PROTOBUF_LIBRARY` and `PROTOBUF_INCLUDE_DIR` instead of `PHOTOBUF_LIBRARY` and `PHOTOBUF_INCLUDE_DIR`? :stuck_out_tongue:

The build script *should* work too... The "no input files" error typically means that no .cpp files were provided to a build target. It can be debugged by printing the lists of input files in the CMakeLists scripts (in this case, probably that of the _CuraEngine library target of CuraEngine's CMakeList).

iashwinprasada, 2015-12-02:
@fieldOfView yeah! i have only modified the GUI as of now, how can i bypass the build step for CuraEngine from cmake? also i just take precomiled binary and put it inside cura-build-master\inst\bin?

i've spent quite a lot of time on this already, but really need this! thanks for replying to my queries :)

fieldOfView, 2015-12-02:
I don't use cmake at all; I am just interested in running Cura directly from source. But I guess you would need to build an actual executable eventually?

Anyways, here's my recipe, which I used this morning on a Win10 machine.

I use the following binaries:
* python3.4 (from https://www.python.org/downloads/release/python-340/)
* pyqt5 (from https://riverbankcomputing.com/software/pyqt/download5)
* numpy (from http://www.lfd.uci.edu/~gohlke/pythonlibs/#numpy)
* protobuf (from https://github.com/GoogleCloudPlatform/gcloud-python-wheels/blob/master/wheelhouse/protobuf-3.0.0a3-py3-none-any.whl)
The .whl files can be installed with pip

You need to download/clone the following repositories in a folder:
Ultimaker
 | Cura
 | Uranium
 | libArcus

In the same folder, add a new folder named bin, containing the following files from a recent Cura build:
* CuraEngine.exe
* libgcc_s_seh-1.dll
* libstdc++-6.dll
* libwinpthread-1.dll

Finally, add these folders to your PYTHONPATH variable:
[repofolder]\libArcus\Python
[repofolder]\Uranium

nallath, 2015-12-02:
Building the executable is only required if you want to distribute it. If you just want to make & test changes there is no need what so ever.

iashwinprasada, 2015-12-02:
Yes! I totally understand that..we are gonna be distributing it with our
printers and hence the need to "build" the package

nallath, 2017-07-27:
Closing due to inactivity.

closed 2017-07-27 by nallath