Ultimaker/Cura #535: HELP !!! Cura 15.06 later open crash

shawnlintw, 2015-11-12:
Hi,

I've install the Cura 15.06.03 from Ultimaker offical site since two weeks ago.
It works fine.
However, I got the error message this morning.

Here is the message from cura >>>

Version: 15.09.83
Platform: Windows-7-6.1.7601-SP1
Qt: 5.4.2
PyQt: 5.4.2

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 16, in <module>
  File "C:\dev\cura-build\build\inst\lib\python3.4\site-packages\UM\Application.py", line 273, in getInstance
  File "C:\dev\cura-build\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 64, in __init__
  File "C:\dev\cura-build\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 55, in __init__
  File "C:\dev\cura-build\build\inst\lib\python3.4\site-packages\UM\Application.py", line 69, in __init__
  File "C:\dev\cura-build\build\inst\lib\python3.4\site-packages\UM\Preferences.py", line 70, in readFromFile
  File "C:\Python34\lib\_collections_abc.py", line 498, in __iter__
    def __iter__(self):
  File "C:\Python34\lib\configparser.py", line 1204, in __getitem__
    return self._parser.get(self._name, key)
  File "C:\Python34\lib\configparser.py", line 773, in get
    d)
  File "C:\Python34\lib\configparser.py", line 374, in before_get
    self._interpolate_some(parser, option, L, value, section, defaults, 1)
  File "C:\Python34\lib\configparser.py", line 423, in _interpolate_some
    "found: %r" % (rest,))
configparser.InterpolationSyntaxError: '%' must be followed by '%' or '(', found: "%20Files%20(x86)/Cura%2000.00\\xff\\xff\\xff\\xff\\x00\\x00\\x00\\x00'"

Ghostkeeper, 2015-11-12:
This looks like a duplicate of #346, #389, #428, #454, #458, #476 and #489. Back then, Cura created corrupt configuration files when the Python environment is interrupted while it is writing that file, or when multiple instances of Cura are writing to it at the same time. This issue is fixed in more recent versions (15.09.85 I think, somewhere around then). You can download a newer version from http://software.ultimaker.com/Cura_closed_beta/15.10/ .

To fix your Cura version, delete the configuration file that gives the error. It's probably in `C:\Users\<you>\AppData\Local\cura\settings`. Cura will then create it anew, but you will lose those configuration settings. You could also try and open that file with a text editor to try to mend it, if you're feeling adventurous and really don't want to lose your settings.

awhiemstra, 2015-11-12:
Additionally, this has been fixed in a more recent build.

closed 2015-11-12 by awhiemstra