Ultimaker/Cura #539: Wish - PauseAtHeight with number of layer and time - christmas is near ...

CaJoDo, 2015-11-14:
for the plugin I have some important wishes:
1. pause at layer. alternatively input the layer where to stop. And the printer will BEEP BEEP BEEP ... when the printer is waiting for me.

2. while displaying the model in "Layers" please show the time, which is gone until this layer will be printed.

3. in the printer display please show the z-Position in millimeters AND in Layers(of the actual Position).
4.showing the time, since print-job-start was started 
 
This will be great.

nallath, 2015-11-16:
The printer display is a firmware issue. There are modified firmwares out there that do this (atleast for the UM)

There is a pause at layer script. It uses the post processing plugin (you can find it in my repo's). The beeping is a firmware thing, we can not control that from Cura.

fieldOfView, 2015-11-16:
For convenience:
https://github.com/nallath/PostProcessingPlugin/

In https://github.com/nallath/PostProcessingPlugin/blob/master/scripts/PauseAtHeight.py you can see that the height at which the pause occurs is set in mm. I think it may be worthwhile to change that to a layer number. That way you can use the Layer View to inspect the exact layer at which the pause occurs. Alternatively, the Layer View could show the height (as well as the layer number?). 

Does the UM support the M300 gcode to "beep"? Not that I think the beep would make much of a difference; the printer going silent is more telling of a pause than a single beep.

Ghostkeeper, 2015-11-16:
When computing the layer number, keep in mind that the first layer may have a different height than the rest.

Vandrasc, 2018-04-09:
This ticket is from 2015, and unfortunately, it is not in our planning to address this issue any time soon. Can we close it? The Pause at layer is there though :)

closed 2018-04-10 by ianpaschal