Ultimaker/Cura #542: remove material after print

CaJoDo, 2015-11-14:
this will be a very very important option. REMOVE MATERIAL AFTER PRINT (as a option with checkbox).
 At the moment I had to reheat the head with "change material" and cancel it, after material is removed.

If I don't remove my PLA, it will break over night (several times).

Ghostkeeper, 2015-11-16:
This could also be implemented as a post processing plug-in with a retract. There is currently no g-code for material change, as far as I'm aware, so to implement removing of material after print would probably require a firmware change of the printer you're using.

BagelOrb, 2017-07-21:
A gcode for removing material could be quite interesting, but I think it always requires user interaction.
Calling @daid for answers

Ghostkeeper, 2017-07-28:
UM3 does this in firmware. All other printers I know of just let it stay where it ends up.

We wanted a feature like this when we implemented that a filament was cooling down when it's no longer necessary, remember? It would cool down the filament after its last layer was printed, and then the UM3 would heat it up again automatically at the end of a print just to move the filament 3mm higher to the parking position...

BagelOrb, 2017-07-28:
I mean to remove filament totally from the whole bowden tube - not remove filament from the hot zone of the bowden tube only...

Ghostkeeper, 2017-08-14:
Ah yeah. That is quite interesting. Depends on the printer, I suppose. Some printers (such as UM3) require the user to indicate that the filament is now indeed gone, which allows new filament to be loaded and the sync with Cura to update properly, etc. Some printers (such as UMO) don't.

Ghostkeeper, 2017-11-17:
Our project manager just removed this from our planning because it's more than 12 weeks old.

ianpaschal, 2018-05-24:
As part of our new ticket handling strategy, we voted as a team on the likelihood of fixing this issue. Unfortunately, we’ve decided that we will not take time to fix it, and are marking it as `Status: Won’t Fix/Do` and closing the issue.

closed 2018-05-24 by ianpaschal