Ultimaker/Cura #544: An uncaught exception has occurred!

Niketin, 2015-11-16:
Hey! This happens when I try to launch Cura.

Version: 15.06.03
Platform: Windows-7-6.1.7601-SP1
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 56, in readFromFile
  File "c:\dev\uranium\UM\Preferences.py", line 122, in _loadFile
  File "c:\dev\python34\lib\configparser.py", line 672, in read
  File "c:\dev\python34\lib\configparser.py", line 1070, in _read
configparser.DuplicateOptionError: While reading from 'C:\\Users\\user/AppData/Local/cura\\cura.cfg' [line 18]: option 'show_overhang' in section 'view' already exists

Ghostkeeper, 2015-11-17:
This is a duplicate of #346, #389, #428, #454, #458, #476 and #535. Version 15.06 created corrupt configuration files when the Python environment is interrupted while it is writing that file, or when multiple instances of Cura are writing to it at the same time. This issue is fixed in version 2.0.

To fix your Cura version, delete the file mentioned in your traceback: C:\Users\user\AppData\Local\cura\cura.cfg. Cura will then create it anew, but you will lose your configuration settings. You could also try and open that file with a text editor such as Notepad to try to mend it, if you're feeling adventurous and really don't want to lose your settings.

closed 2015-11-17 by Ghostkeeper

fieldOfView, 2015-11-17:
Note: "version 2.0" is the new "version 15.10"