Ultimaker/Cura #545: Multiple Objects -- only one has a first layer, the rest start at layer 2, floating

nebarnix, 2015-11-16:
Hi. When I load a bed full of objects, the first one loaded gets the skirt and the first layer, and the rest 'float' up at layer 2 with the fans on. The print fails spectacularly.

[image]

[image]

[image]

See them floating?!
[image]

fieldOfView, 2015-11-17:
Note that this is legacy-cura, which is hosted here (but won't get any new development): https://github.com/daid/Cura
If you could check if this still happens with the recent betas, that would be great.
http://software.ultimaker.com/Cura_closed_beta/15.10/

BagelOrb, 2017-07-21:
Problem persists in current Cura.
Solution is to pass drop_on_buildplate to the engine and enforce it per model.

Ghostkeeper, 2017-12-15:
This is fixed for Cura 3.2.

closed 2017-12-15 by Ghostkeeper