Ultimaker/Cura #546: setMachineSettingValue function in MachineManagerProxy.py doesn't trigger reslice

mpbomil, 2015-11-17:
I'm currently trying to write a plugin for Cura. 
When I use the function [setMachineSettingValue [line 119]](https://github.com/Ultimaker/Uranium/blob/7150ee94276782705e72aec0b705057bf24c66aa/UM/Qt/Bindings/MachineManagerProxy.py#L119) to change my printer settings --
I expect that a reslice will occur and the GCode will be updated. 
This is not the case though, the GCode doesn't renew; it still holds the previous settings.
*(e.g. machine_gcode_flavor, machine_start_gcode or machine_end_gcode)*

**Compare that with the function that changes the *profile* setting:**
When a *profile* setting is changed by using
`def setSettingValue(self, key, value):` [line. 74](https://github.com/Ultimaker/Uranium/blob/7150ee94276782705e72aec0b705057bf24c66aa/UM/Settings/Profile.py#L74)

settingValueChanged is emitted:
`self.settingValueChanged.emit(key)` [line. 88](https://github.com/Ultimaker/Uranium/blob/7150ee94276782705e72aec0b705057bf24c66aa/UM/Settings/Profile.py#L88)

The slicer listens to the emit with:
`self._profile.settingValueChanged.disconnect(self._onSettingChanged)` [line. 174](https://github.com/Ultimaker/Cura/blob/master/plugins/CuraEngineBackend/CuraEngineBackend.py#L174)

This results in the reslice being triggered:
`self._change_timer.start()` [line. 240](https://github.com/Ultimaker/Cura/blob/master/plugins/CuraEngineBackend/CuraEngineBackend.py#L240)

 
**Question:** 
Why does changing the profilesettings trigger a reslice, but a change in the machinesettings won't?

Here's the function that changes the machineSetting:

    def setMachineSettingValue(self, setting, value):
        if not self._machine_definition.isMachineSetting(setting):
            Logger.log("w", "Tried to override setting %s that is not a machine setting", setting)
            return

        self._machine_setting_overrides[setting] = value

nallath, 2015-11-17:
Setting machineSetting values should trigger a re-slice. I think we just overlooked this (as we don't change machine settings during runtime)

mpbomil, 2015-11-17:
Would it be possible to change machine settings *during* runtime? Since that is what I'm trying to achieve.

nallath, 2015-11-18:
I think it should be possible. Nozzle size is one of the things that's in the machine definitions and that is something that could be upgraded.

mpbomil, 2015-11-18:
Is it true that the .cfg files in C:\Users\<user>\AppData\Local\cura\machine_instances do **not** overwrite the machine settings?

	[general]
	name = Ultimaker 2
	type = ultimaker2
	version = 1

	[machine_settings]
	machine_gcode_flavor = RepRap (Marlin/Sprinter)

This cfg for example does not overwrite the default UltiGCode with RepRap. I can see this by doing:
`getattr(Application.getInstance().getController().getScene(), "gcode_list")`
and looking at the gCode result I get, it's still on ultiGCode even after reslicing.

awhiemstra, 2015-11-18:
It should. Currently the functionality is used for the heated bed upgrade for the UMO which works as far as I know.

In regards to the original issue, it mostly is just an oversight. Setting a machine setting value should also emit a signal that that value is changed. We just had no real need for it so far.

BagelOrb, 2017-07-21:
Probably fixed in the mean time.

closed 2017-07-21 by BagelOrb