Ultimaker/Cura #548: Can't get GCode to update after changing machine settings through code

mpbomil, 2015-11-18:
1. I use this code to set the machine_gcode_flavor from UltiGCode to RepRap:
`Application.getInstance().getMachineManager().getActiveMachineInstance().setMachineSettingValue("machine_gcode_flavor","RepRap (Marlin/Sprinter)")`

2. Afterwards, I use getMachineSettingValue to see if it changed it:
      `Application.getInstance().getMachineManager().getActiveMachineInstance().getMachineSettingValue("machine_gcode_flavor")`
Which returns the value of RepRap, like it should.

3. Then I force a reslice by using:
`Application.getInstance().getBackend().forceSlice()`

4.  I log the gcode_list, to see if the flavour has changed:
`getattr( Application.getInstance().getController().getScene(), "gcode_list")`

5. But it hasn't:
`[';FLAVOR:UltiGCode', ';TIME:34', ';MATERIAL:51', ';MATERIAL2:0', '', ';Generated with Cura_SteamEngine DEV', ';LAYER_COUNT:8', ';LAYER:0', 'M107', 'G0 F9000.000000 X51.662 Y54.436 Z0.299', 'G0 X55.020 Y53.078', ';TYPE:SKIRT', 'G1 F900.000 X56.328 Y53.053 E0.15647', 'G1 X56.584 Y53.054 E0.18708', 'G1 X57.192 Y53.063 E0.25981', 'G1 X57.701 Y53.087 E0.32075', 'G1 X57.941 Y53.106 E0.34955', 'G1 X57.943 Y53.106 E0.34979'...`

I've been trying to figure this out for a long time now, can anybody shed light on this?

mpbomil, 2015-11-23:
Can anybody help me out with this issue?

nallath, 2015-11-25:
Could be that only the comment isn´t changed correctly. Can you see if it uses the correct type for the code? 
Could you also check the engine logs to see if it gave any errors?

awhiemstra, 2015-11-25:
You could try logging what is sent to the engine to verify that everything is sent correctly. Though I see no reason why it would not be sent correctly. If everything is sent ok, the next step would be checking to see what the engine does with the setting.

closed 2015-11-27 by awhiemstra