Ultimaker/Cura #549: Heated bed setting enabled, but not displayed

CCS86, 2015-11-21:
[image]

CCS86, 2015-11-21:
11/17 build

CCS86, 2015-11-21:
Same with support line width:

[image]

Ghostkeeper, 2015-11-23:
Perhaps you are using a machine without heated bed, such as Ultimaker Original (causing `heated_bed_enable` to be false) and have support disabled (causing `support_enabled` to be false)?

This sort of thing calls for a thought on how visibility is inherited.

CCS86, 2015-11-23:
You were 100% correct on the support setting, and probably on the heated bed as well.

On the bed, this should be an easy fix no? I don't see a way to edit machine instances.

CCS86, 2015-11-23:
I was able to create a new machine instance that enabled the heated bed. It seems like editing an existing instance would be helpful too.

Ghostkeeper, 2015-11-24:
I'm glad you got it working for yourself. We aren't about to let the user edit machines through Cura. Those that modify their machines to have extra features such as a heated bed will most likely not be taken aback by modifying a JSON file either, such as yourself. The rest of the users with uncommon printers may be satisfied with the profiles we already have, such as the UMO+ for UMO printers with heated bed.

Making the visibility of settings with respect to machine instances and setting inheritance work better would probably be a simple fix in code, but requires some thought for user experience.

fieldOfView, 2015-11-24:
> Those that modify their machines to have extra features such as a heated bed will most likely not be taken aback by modifying a JSON file either, such as yourself.

The official Heated Bed Kit comes with very nice documentation, making it fairly straightforward to add the heated bed. The documentation could include similarly straightforward instructions to edit the json file.

awhiemstra, 2015-11-24:
Actually, the options that are available in the add machine wizard should also be editable from the printer management page. However, only those options should be editable. This specifically includes the heated bed option.

CCS86, 2015-11-29:
For some reason, the bed temp field has disappeared again. Here is my machine instance settings:

[general]
name = Ultimaker Original
type = ultimaker_original
version = 1

[machine_settings]
machine_heated_bed = True
machine_extruder_drive_upgrade = True

CCS86, 2015-12-19:
Were you guys able to see where this issue lies?

[image]

[image]

Even with it enabled, no bed on gcode is generated.

Ghostkeeper, 2015-12-21:
No, we haven't looked at it yet as far as I know. But the problem can be reproduced on this end too.

Ghostkeeper, 2015-12-21:
I see now that the issue in our internal issue tracker is scheduled for the release after the one we're working on now, so it's going to take a while.

CCS86, 2015-12-22:
Thanks for the update. I was hoping it was a simple issue to solve. Not a massive issue to work around, but an annoyance for sure.

CCS86, 2016-02-23:
On the 2/20 build, I still am having this issue. I tried creating a new printer, selected UMO, checked boxes for heated bed and drive upgrade, but still cannot see bed temp in my settings.

CCS86, 2016-02-23:
Here is the CFG file Cura created for the new machine (in machine instances):

[general]
name = Ultimaker Original ``
type = ultimaker_original
active_profile = Normal Quality
version = 1
material = PLA

[machine_settings]
machine_heated_bed = True
machine_extruder_drive_upgrade = True

nallath, 2016-02-23:
Ugh. This is caused by a hack that I made to ensure that settings that can never be enabled are hidden in settings view. I hacked that together by checking if enabled is "false" at startup and then setting a 'never enabled' flag. However that assumption is false in thise case, as the machine instance overrides that.

I'll log another issue in our internal tracker.

CCS86, 2016-03-08:
Is there any workaround I can do for now?

Thanks

fieldOfView, 2016-03-08:
Unfortunately we haven't quite found the definitive fix yet (it is not on the list for the 2.1 release).

As a workaround, you could edit fdmprinter.json to just always enable the "material_bed_temperature" setting (set "enabled" to "True" at around line 691). 

Eventually we will get to the bottom of this!

CCS86, 2016-03-08:
Just in case this helps, on line 58, I changed this default from false to true and it now shows up in my settings:

`        "machine_heated_bed": {
            "description": "Whether the machine has a heated bed present.",
            "default": true,
            "global_only": true`

Ghostkeeper, 2016-03-09:
That helps, because I think that default shouldn't have been used, since the printer should overwrite that default in fdmprinter.json.

This is in our internal issue tracker (CURA-1041) the seventh in the priority list for version 2.2. I think we'll get around to it somewhere next week.

fieldOfView, 2017-06-12:
This has been fixed, issue can be closed

closed 2017-06-13 by awhiemstra