Ultimaker/Cura #550: How does cura15.10 has "printer with USB" function ?

shawnlintw, 2015-11-22:
Hi there,

I am install the cura 15.10 version on my PC.
OS : Windows 7 64 bit.  There is a COM port on my computer (COM 1).
FDM : Prusa i3 

Usually, I plug in the FDM via USB connection. 
I has checked the connection status, FDM using the  COM port 3 to communication.
The driver has installed. It works good in Cura 15.04 on same computer.

The question is I can't see any "printer with USB" button as when I change Cura to version 15.10.
Has anything I forget did ?
Could anyone helps ?

Ghostkeeper, 2015-11-23:
Perhaps Cura 15.04 is still running? An instance of Cura tends to hog the USB connection for itself. The new Cura 15.10 checks for available USB connections at startup.

nallath, 2015-11-25:
Post the logs. It should mention something about trying to connect with certain ports (and how well that went)

closed 2017-07-21 by BagelOrb