Ultimaker/Cura #551: Verbose GCode

CCS86, 2015-11-22:
Is there a reason that Cura isn't posting out GCode which includes all settings? It seems like such a small physical size, for so much valuable diagnostic information. I appreciate that feature even for personal use, but especially in a beta development cycle, I can't imagine it being omitted!

Ghostkeeper, 2015-11-23:
There was talk of this a while ago. Also in relation to the 3MF file format which allows some settings to go along with the 3D model.

BagelOrb, 2017-07-21:
I still think it would be nice to include all flattened settings in the gcode.
Could be nice to identify problems in gcode, whether it's a user setting problem, a setting calculation error or a slicing error.

Ghostkeeper, 2017-12-04:
I made a ticket to do that (save all settings instead of just the ones that are not saved to a profile). But it got rejected by our project manager for being older than 12 weeks.

ianpaschal, 2018-05-24:
The reason for this is because we find it vastly more useful to see which settings have been changed when analyzing print quality. Including all settings would actually make this more difficult so we’ve decided that we will not take time to fix it, and are marking it as `Status: Won’t Fix/Do` and closing the issue.

closed 2018-05-24 by ianpaschal