Ultimaker/Cura #555: dev 20151117 gcode for minimal layer time 1 and 8 sec are identical

EugeneSkrebnev, 2015-11-25:
dev 20151117

slicing cube 1x1x1
changing minimal layer time - 
got 2 identical files

https://gyazo.com/2bf529d363651180e6657538ca0d5c9d
my printer profile:
https://www.dropbox.com/s/xdpz940m4qy6b8z/Eugene%27s%203D%20Printer.json?dl=0

EugeneSkrebnev, 2015-11-25:
in 15.04.2 with speed 50(the same as new) and minimal layer time 8sec it prints pretty slower

Ghostkeeper, 2015-11-26:
I tried slicing with your profile in the latest Cura, with a 1cm wide cube.

It produces 50 layers. With the default minimal layer time of 5s it estimates it takes 5 minutes to print. With a minimal layer time of 8s it estimates it takes 7 minutes to print. Though I didn't check the g-code yet, it seems that the g-code is different on my computer.

BagelOrb, 2017-07-21:
I can't imagine this is still a problem

closed 2017-07-21 by BagelOrb