Ultimaker/Cura #556: How change nozzle size in new Cura (15.09 Or 1.99)?

nacho-pulido, 2015-11-26:


Ghostkeeper, 2015-11-27:
The setting is called Line Width, in the category Quality.

closed 2015-11-27 by nallath

nacho-pulido, 2015-11-27:
ok, but there are a lot of "x width", I have noted that when I changed the line width value, the other width paremeters remain with previous value, is necessary change all? There is any number for thes seconday parameters that adopt the general value?

Ghostkeeper, 2015-11-30:
The wall line width, outer wall line width, other walls line width, skirt line width, top/bottom line width, infill line width and support line width all inherit from the line width. That means that if you change the line width, the "x line width" settings should also change.

If you change any of the "x line width" settings, the changed settings will stay put even if you change the global line width, since the inheritance only applies to the default (so if you change the "x line width" setting it is no longer default).

If this is not happening in the way I described, you've found a bug! Some bugs regarding inheritance were fixed two weeks ago, so if you're using an older version, it's a good idea to [update](https://software.ultimaker.com/Cura_closed_beta).