Ultimaker/Cura #557: Duplicating objects frequently crashes front end

CCS86, 2015-11-27:
With the 11-17 build, I frequently get "cura stopped working" crashes when duplicating items.

Ghostkeeper, 2015-11-30:
That's pretty serious. Thanks for the report! I'm going to assume you're using Windows since that appears in a log in issue #559.

CCS86, 2015-11-30:
Sorry for not specifying! It is Windows 10.

Ghostkeeper, 2015-12-22:
I finally came around to debugging this but could not reproduce it, neither on my Linux computer nor on the Windows 10 computer.

Perhaps it has been fixed in a more recent build. Perhaps the bug has been hidden so that it won't come to light any more. In any case, I'm hopeful that you won't get the crash any more once I've managed to get you that win32 build.

BagelOrb, 2016-01-07:
Perhaps it depends on the kind of object you are duplicating. If it happens with some models more than others, can you provide any of us with an STL file?

Ghostkeeper, 2016-01-13:
@CCS86 If you contact me at r.dulek (at) ultimaker (.) com I can provide you with a Win64 build. There are no secrets in this build but since the public beta closed we can't publish it publicly due to its instability vs. 15.04. The build includes all the latest features, including UM2+ and some extra parameters. Would you like to try it?

In exchange, we'd like you to try and reproduce this issue since we aren't able to reproduce it on this end.

daid, 2016-01-13:
Could this be an out of virtual memory issue?

closed 2016-12-26 by thopiekar