Ultimaker/Cura #559: 'getSettingValue' Log Errors

CCS86, 2015-11-29:
Not sure how important these are, but thought I would report it anyway:

[cura.log.txt](https://github.com/Ultimaker/Cura/files/46781/cura.log.txt)

Ghostkeeper, 2015-11-30:
It's almost like your machine setting files are corrupt.

nallath, 2015-11-30:
Did you manually change the definition file?

CCS86, 2015-11-30:
I have not modified the machine configuration file, which is in the install directory.

The machine instance file in App data is pretty simple:

[general]
name = Ultimaker Original
type = ultimaker_original
version = 1

[machine_settings]
machine_extruder_drive_upgrade = True
machine_heated_bed = True

Ghostkeeper, 2017-07-27:
Doesn't reproduce for me.

BagelOrb, 2017-07-28:
I think you actually reopened instead of closed, because it doesn't reproduce for you.
And also this ticket is from 2015.

closed 2017-07-28 by BagelOrb