Ultimaker/Cura #561: Crash on startup

graphics80, 2015-12-01:
If I start Cura 15.06.03 on my Win7 64bit. It starts, as soon as I choose a Printer-Model (Ultimaker 2, Ultimaker 2 extended, Ultimaker 2 Go, Ultimaker Original+) it crashes. 
Ultimaker Original works.

When I choose Original at first start Up, I can add a Ultimaker 2 extended or other afterwards. But when I select then Machine > Ultimaker 2 Extended Cura crashes again...

Nothing is written in the Engine Log

cura.log
------------------------------------------------
2015-12-01 13:09:28,255 - INFO - Loaded plugin FileLogger
2015-12-01 13:09:28,374 - INFO - Loaded plugin LocalFileStorage
2015-12-01 13:09:28,379 - INFO - Loaded plugin LayerView
2015-12-01 13:09:28,388 - INFO - Loaded plugin MeshView
2015-12-01 13:09:28,395 - INFO - Loaded plugin WireframeView
2015-12-01 13:09:28,398 - INFO - Loaded plugin OBJReader
2015-12-01 13:09:28,400 - INFO - Loaded plugin STLReader
2015-12-01 13:09:28,404 - INFO - Loaded plugin STLWriter
2015-12-01 13:09:28,406 - INFO - Loaded plugin GCodeWriter
2015-12-01 13:09:28,411 - INFO - Loaded plugin CameraTool
2015-12-01 13:09:28,416 - INFO - Loaded plugin MirrorTool
2015-12-01 13:09:28,434 - INFO - Loaded plugin RotateTool
2015-12-01 13:09:28,444 - INFO - Loaded plugin ScaleTool
2015-12-01 13:09:28,452 - INFO - Loaded plugin SelectionTool
2015-12-01 13:09:28,457 - INFO - Loaded plugin TranslateTool
2015-12-01 13:09:28,466 - INFO - Checking for new version of cura
2015-12-01 13:09:28,466 - INFO - Loaded plugin UpdateChecker
2015-12-01 13:09:28,480 - INFO - Loaded plugin USBPrinting
2015-12-01 13:09:28,481 - INFO - Loaded plugin CuraEngineBackend
2015-12-01 13:09:29,159 - DEBUG - Loaded theme C:/Program Files (x86)/Cura_15.06.03/resources/themes/cura
2015-12-01 13:09:29,359 - INFO - Started engine process: C:\Program Files (x86)\Cura_15.06.03\CuraEngine.exe
2015-12-01 13:09:29,898 - DEBUG - Backend connected on port 49674

dxDiag
---------------------------------------------------------------------------------------------------------------
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/48485/DxDiag.txt)

fieldOfView, 2015-12-01:
Please try one of the more recent builds of the open beta instead of 15.06.03:
https://ultimaker.com/en/community/17296-cura-1510-open-beta
A lot has happened since 15.06.03

graphics80, 2015-12-01:
Just tried with dev20151117, same Issue...

ripper121, 2015-12-21:
I had also Startup Problem but with config file.

Version: 15.06.03
Platform: Windows-7-6.1.7601-SP1
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 56, in readFromFile
  File "c:\dev\uranium\UM\Preferences.py", line 122, in _loadFile
  File "c:\dev\python34\lib\configparser.py", line 672, in read
  File "c:\dev\python34\lib\configparser.py", line 1088, in _read
configparser.ParsingError: Source contains parsing errors: 'C:\\Users\\sts/AppData/Local/cura\\cura.cfg'
	[line 17]: 'end]\n'
	[line 19]: ' 2\n'

nallath, 2015-12-21:
@ripper121; This is an issue with 15.06 specifically. Upgrading should resolve it.

BagelOrb, 2017-07-21:
outdated

closed 2017-07-21 by BagelOrb