Ultimaker/Cura #565: Crash on start

giutrec, 2015-12-09:
Hi, I'm one ArchLinux user, i've tried to install the cura-git package (and the dependencies), but on start cura crash and show this message
Version: master
Platform: Linux-4.2.5-1-ARCH-x86_64-with-arch
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/cura/CuraSplashScreen.py", line 25, in drawContents
    painter.drawText(0, 0, 203, 255, Qt.AlignRight | Qt.AlignBottom, version[1])
IndexError: list index out of range

regards
Giuseppe

fieldOfView, 2015-12-09:
I fixed this in the feature_ux2-1 branch a couple of days ago. I hope the PR gets merged soon.

Easiest fix for now is to comment out line 25 of CuraSplashScreen.py

Ghostkeeper, 2015-12-09:
I get the message too but it still launches properly.

fieldOfView, 2015-12-09:
The code in CuraSplashScreen blindly tries to split the version number in major/minor, but since the master branch has its version set to "master" instead of eg "2.0", this fails. The offending code can be safely commented out if it prevents launch.

fieldOfView, 2015-12-09:
@daid committed a proper fix: https://github.com/Ultimaker/Cura/commit/3d4b626c50f7013fea729b1c3da4f0ba5cac92e8

closed 2015-12-09 by fieldOfView

daid, 2015-12-09:
Note, for me Cura did start with this exception happening. On Windows that is.