Ultimaker/Cura #566: Cant place duplicated object

EugeneSkrebnev, 2015-12-11:
If there is no space after duplicating object. cura cant select place for it.
see gif file below:
https://gyazo.com/3ea39b5a1234f62863d3fcaf2e4773a0

nallath, 2015-12-14:
Switch to 1.99. You can find it on http://software.ultimaker.com/Cura_closed_beta/15.10/

EugeneSkrebnev, 2015-12-14:
Cura-1.99.0-dev.20151117-Darwin.dmg	17-Nov-2015 14:14	54M	 
it was this version. i already use Cura_closed_beta builds

as i can see its the latest one

nallath, 2015-12-14:
Odd, this should have been fixed in that version.

EugeneSkrebnev, 2015-12-14:
it was easy to reproduce. just add alot of small objects about 1 by 1 sm in (xy) and after that start duplicating them using right click - duplicate. at first them start jumping as in checkers game,
and after few iterations it fails with infinity loop.

Ghostkeeper, 2015-12-14:
Checkers is a very good way to describe it :stuck_out_tongue:

BagelOrb, 2017-07-21:
outdated

closed 2017-07-21 by BagelOrb

Ghostkeeper, 2017-07-27:
Fixed in the meanwhile.