Ultimaker/Cura #569: Cura.log  Error  ask for help!!

maoshenglinTw, 2015-12-18:
2015-12-18 14:50:27,222 - INFO - Loaded plugin ConsoleLogger
2015-12-18 14:50:27,432 - INFO - Loaded plugin FileLogger
2015-12-18 14:50:27,794 - INFO - Loaded plugin LocalFileStorage
2015-12-18 14:50:27,804 - INFO - Loaded plugin LayerView
2015-12-18 14:50:27,824 - INFO - Loaded plugin MeshView
2015-12-18 14:50:27,834 - INFO - Loaded plugin WireframeView
2015-12-18 14:50:27,844 - INFO - Loaded plugin OBJReader
2015-12-18 14:50:27,844 - INFO - Loaded plugin STLReader
2015-12-18 14:50:27,854 - INFO - Loaded plugin OBJWriter
2015-12-18 14:50:27,854 - INFO - Loaded plugin STLWriter
2015-12-18 14:50:27,864 - INFO - Loaded plugin GCodeWriter
2015-12-18 14:50:27,874 - INFO - Loaded plugin CameraTool
2015-12-18 14:50:27,894 - INFO - Loaded plugin MirrorTool
2015-12-18 14:50:27,914 - INFO - Loaded plugin RotateTool
2015-12-18 14:50:27,934 - INFO - Loaded plugin ScaleTool
2015-12-18 14:50:27,944 - INFO - Loaded plugin SelectionTool
2015-12-18 14:50:27,954 - INFO - Loaded plugin TranslateTool
2015-12-18 14:50:27,964 - ERROR - Error loading plugin PostProcessingPlugin:
2015-12-18 14:50:27,964 - ERROR - 'CuraApplication' object has no attribute 'getOutputDeviceManager'
2015-12-18 14:50:27,974 - INFO - Checking for new version of cura
2015-12-18 14:50:27,974 - INFO - Loaded plugin UpdateChecker
2015-12-18 14:50:28,004 - INFO - Loaded plugin USBPrinting
2015-12-18 14:50:28,014 - INFO - Loaded plugin CuraEngineBackend
2015-12-18 14:50:28,299 - INFO - Started engine process: C:\Program Files (x86)\Cura 15.06\CuraEngine.exe
2015-12-18 14:50:31,093 - DEBUG - Loaded theme C:/Program Files (x86)/Cura 15.06/resources/themes/cura
2015-12-18 14:50:32,763 - DEBUG - Backend connected on port 49674

2015-12-18 14:50:27,964 - ERROR - Error loading plugin PostProcessingPlugin:
2015-12-18 14:50:27,964 - ERROR - 'CuraApplication' object has no attribute 'getOutputDeviceManager'

What's that mean?
Is anyone can answer to me!!   Thanks a lot!

maoshenglinTw, 2015-12-18:
environment: Win7 64 bit

Ghostkeeper, 2015-12-18:
Seems that http://github.com/nallath/PostprocessingPlugin is behind the times. Fault on our end (or @nallath).

You can temporarily fix it by removing the plug-in from Cura's plug-in directory.

maoshenglinTw, 2015-12-18:
Thanks a a lot!   My Cura is not up-to-date version.

closed 2015-12-18 by Ghostkeeper

nallath, 2015-12-18:
Did you use the right version of the post processing plugin? The repo has multiple branches for the different Cura versions.