Ultimaker/Cura #573: [15.10] OpenGLESv2  used instead of regular OpenGL

thopiekar, 2015-12-21:
As far as I know Cura is currently just supporting the regular OpenGL.
I just built Cura for Ubuntu armhf (have a Odroid U3 board here) and it seems that on armhf OpenGLESv2 is used by Qt5 as default, so OpenGL should be forced then.

```
thopiekar@gate:~$ export LANG=C
thopiekar@gate:~$ cura_app.py 
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/cura/CuraSplashScreen.py", line 25, in drawContents
    painter.drawText(0, 0, 203, 255, Qt.AlignRight | Qt.AlignBottom, version[1])
IndexError: list index out of range
QPainter::end: Painter ended with 2 saved states
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/cura/CuraSplashScreen.py", line 25, in drawContents
    painter.drawText(0, 0, 203, 255, Qt.AlignRight | Qt.AlignBottom, version[1])
IndexError: list index out of range
QPainter::end: Painter ended with 2 saved states
2015-12-21 13:50:20,840 - INFO - Loaded plugin ConsoleLogger
2015-12-21 13:50:20,862 - INFO - Loaded plugin FileLogger
2015-12-21 13:50:20,898 - INFO - Loaded plugin LocalFileOutputDevice
2015-12-21 13:50:20,918 - INFO - Loaded plugin WireframeView
2015-12-21 13:50:20,943 - INFO - Loaded plugin MeshView
2015-12-21 13:50:20,989 - INFO - Loaded plugin MirrorTool
2015-12-21 13:50:21,016 - INFO - Loaded plugin SelectionTool
2015-12-21 13:50:21,051 - INFO - Loaded plugin TranslateTool
2015-12-21 13:50:21,099 - INFO - Loaded plugin RotateTool
2015-12-21 13:50:21,122 - INFO - Loaded plugin CameraTool
2015-12-21 13:50:21,153 - INFO - Loaded plugin PerObjectSettingsTool
2015-12-21 13:50:21,196 - INFO - Loaded plugin ScaleTool
2015-12-21 13:50:21,218 - INFO - Loaded plugin STLReader
2015-12-21 13:50:21,237 - INFO - Loaded plugin OBJWriter
2015-12-21 13:50:21,259 - INFO - Loaded plugin STLWriter
2015-12-21 13:50:21,282 - INFO - Loaded plugin OBJReader
2015-12-21 13:50:21,283 - WARNING - Plugin ConsoleLogger was already loaded
2015-12-21 13:50:21,342 - INFO - Loaded plugin USBPrinting
2015-12-21 13:50:21,385 - INFO - Loaded plugin RemovableDriveOutputDevice
2015-12-21 13:50:21,579 - INFO - Loaded plugin CuraEngineBackend
2015-12-21 13:50:21,599 - INFO - Loaded plugin LayerView
2015-12-21 13:50:21,645 - INFO - Loaded plugin 3MFReader
2015-12-21 13:50:21,751 - INFO - Loaded plugin SliceInfoPlugin
2015-12-21 13:50:21,771 - INFO - Loaded plugin GCodeWriter
2015-12-21 13:50:21,802 - INFO - Loaded plugin ChangeLogPlugin
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/cura/CuraSplashScreen.py", line 25, in drawContents
    painter.drawText(0, 0, 203, 255, Qt.AlignRight | Qt.AlignBottom, version[1])
IndexError: list index out of range
QPainter::end: Painter ended with 2 saved states
2015-12-21 13:50:21,915 - DEBUG - Setting value of setting layer_height to 0.06
2015-12-21 13:50:21,916 - WARNING - Tried to set value of non-user setting layer_height
2015-12-21 13:50:21,918 - DEBUG - Setting value of setting layer_height to 0.1
2015-12-21 13:50:21,919 - WARNING - Tried to set value of non-user setting layer_height
2015-12-21 13:50:21,921 - DEBUG - Setting value of setting layer_height to 0.08
2015-12-21 13:50:21,922 - WARNING - Tried to set value of non-user setting layer_height
2015-12-21 13:50:21,924 - DEBUG - Setting value of setting layer_height to 0.15
2015-12-21 13:50:21,925 - WARNING - Tried to set value of non-user setting layer_height
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/cura/CuraSplashScreen.py", line 25, in drawContents
    painter.drawText(0, 0, 203, 255, Qt.AlignRight | Qt.AlignBottom, version[1])
IndexError: list index out of range
QPainter::end: Painter ended with 2 saved states
2015-12-21 13:50:21,946 - INFO - Started engine process: /usr/bin/CuraEngine
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/cura/CuraSplashScreen.py", line 25, in drawContents
    painter.drawText(0, 0, 203, 255, Qt.AlignRight | Qt.AlignBottom, version[1])
IndexError: list index out of range
QPainter::end: Painter ended with 2 saved states
2015-12-21 13:50:21,969 - DEBUG - Backend connected on port 49674
2015-12-21 13:50:21,971 - DEBUG - Setting active view to MeshView
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/cura/CuraSplashScreen.py", line 25, in drawContents
    painter.drawText(0, 0, 203, 255, Qt.AlignRight | Qt.AlignBottom, version[1])
IndexError: list index out of range
QPainter::end: Painter ended with 2 saved states
2015-12-21 13:50:24,831 - DEBUG - Loaded theme /usr/share/cura/resources/themes/cura
file:///usr/share/cura/resources/qml/SidebarSimple.qml:203:22: Unable to assign [undefined] to bool
file:///usr/share/cura/resources/qml/SidebarSimple.qml:203:22: Unable to assign [undefined] to bool
file:///usr/lib/arm-linux-gnueabihf/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/arm-linux-gnueabihf/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/arm-linux-gnueabihf/qt5/qml/QtQuick/Controls/Menu.qml:146: TypeError: Cannot read property '__contentItem' of null
file:///usr/lib/python3/dist-packages/UM/Qt/qml/UM/Preferences/MachinesPage.qml:30: TypeError: Cannot read property 'name' of null
file:///usr/lib/python3/dist-packages/UM/Qt/qml/UM/Preferences/MachinesPage.qml:33: TypeError: Cannot read property 'typeName' of null
file:///usr/lib/python3/dist-packages/UM/Qt/qml/UM/Preferences/MachinesPage.qml:35: TypeError: Cannot read property 'hasVariants' of null
file:///usr/lib/python3/dist-packages/UM/Qt/qml/UM/Preferences/MachinesPage.qml:36: TypeError: Cannot read property 'variantName' of null
file:///usr/lib/python3/dist-packages/UM/Qt/qml/UM/Preferences/MachinesPage.qml:36: TypeError: Cannot read property 'hasVariants' of null
file:///usr/lib/python3/dist-packages/UM/Qt/qml/UM/Preferences/MachinesPage.qml:42: TypeError: Cannot read property 'name' of null
file:///usr/lib/python3/dist-packages/UM/Qt/qml/UM/Preferences/MachinesPage.qml:47: TypeError: Cannot read property 'name' of null
Failed to load libGLESv2
Aborted (core dumped)
```

thopiekar, 2015-12-21:
Sorry, I just found the reason while playing with strace.
The reason is that there is a instance while running cura (one of its libraries, pyqt5 or qt5 itself), which is looking for the libGLESv2.so.
Ubuntu and other distributions are using specific folders, in my case it's /lib/arm-linux-gnueabihf/mesa-egl/libGLESv2.so. In case closed-source drivers are installed it can be also /lib/arm-linux-gnueabihf/{xy}/libGLESv2.so. 
The correct method and probably fastest (without browsing non-existing folders I saw in strace) is to implement something like the command `ldconfig -p`, which lists all libraries and their correct paths.

daid, 2015-12-21:
Actually, the windows version of Cura defaults to the OpenGLESv2 bindings by default on my machine, so there is no reason why it cannot use OpenGLESv2

thopiekar, 2015-12-21:
Yes, you are right. I just suggested we are not supporting OpenGLES. Like I
said before there is just something wrong with a function, which is
searching for the libGLESv2.so. It is just not finding the library
correctly.
Creating a symlink from /lib/*/mesa-egl/libGLESv2.so to /lib/*/libGLESv2.so
corrects this problem.
However, if we are not opening /lib/*/libGLESv2.so in our code, it seems to
be related to PyQt5 or Qt5 itself. So I'll go and report this issue at
Ubuntu's bugtracker, of course. ;)

nixomose, 2016-01-02:
Is this why cura hangs when starting up on xubuntu 15.10 but not 14.04? Anybody else see that? The main window comes up but then no interaction with the UI is possible.

thopiekar, 2016-01-02:
@nixomose: Don't know. Are you using Xubuntu on armhf? Or on amd64/i386?
On amd64/i386 the regular OpenGL should be used by default. Which version do you use? How did you install Cura? By "*.deb" or using the code from "2.1" branch?
Additionally, I use my own packaging, hosted in my PPA at launchpad. I just created it to produce daily builds from "master", but now I also have a stable PPA built from 2.1, which should be almost stable ^^

nixomose, 2016-01-03:
I think it must be something else then.
I'm running xubuntu on a 64 bit x86 machine.
I don't have a openglesv2.so, I only have an openglew.so and there's no second version of it.
Interestingly, it seems to be an X thing, because if I ssh -X to the xubuntu 15.10 machine from a xubuntu 14.04 machine to bounce the X connection, it runs just fine.
Not the ideal solution, but workable.

So something is wrong with xubuntu 15.10 that it hangs up when trying to render the platform area.

thopiekar, 2016-01-03:
@nixomose: Well, there should be no openglesv2.so the correct name is
GLESv2.so, I think.
However, as it doesn't seem to be related to this issue, open a new issue
report and add all information about your installation (including *Ubuntu
and Cura). Finally also take your log (your can find it at
"~/.local/share/cura/" named "cura.log"), upload the content to a pastebin
service and add the link to your issue report.

Hope it helps :)

closed 2016-12-26 by thopiekar