Ultimaker/Cura #577: [2.1] Open file dialog: *.gcode not in list of supported files

thopiekar, 2015-12-23:
While trying to open sample gcode provided by bq on their homepage for the Hephestos 2, I noticed that the "open file" has .gcode not in the list of supported files.
Should be shown when the needed plugin is enabled.

Ghostkeeper, 2015-12-24:
Directly loading g-code from file was dropped with the rewrite between version 15.04 and 15.06. It might get re-inserted later but it's currently a regression.

The primary reason it's not in yet is because due to performance issues the layer view currently shows the layers per-layer as flat planes. When showing the g-code such an optimisation is impossible since the g-code of a layer doesn't need to be flat.

fieldOfView, 2016-08-31:
It is not just that the filedialog does not list .gcode as a input format; Cura cannot parse gcode at this point. Layerview is not implemented as a gcode viewer, but it gets data specially prepared/formatted for layerview. This is not a trivial thing to fix.

nallath, 2016-09-01:
I've put a bit of effort into it at some point and it's not that hard. The main problem would be making it fast. You would need to make a GCodeReader plugin that creates polygon data for the layerview to display.

nallath, 2017-07-27:
This is fixed.

closed 2017-07-27 by nallath