Ultimaker/Cura #579: Support for double (or triple) extrusion

Patola, 2015-12-25:
Hi,

I have seen issue #89 and searched carefully in the tracker but I could not find this mentioned.
How is cura 15.10 going to support double extrusion? 15.04 worked perfectly with that and had a neat interface to deal with it.

I am interested in adding a bunch of .json files for some dual-extruder printers, that's why I want to know. I've seen x_offset and y_offset mentioned somewhere but no instructions on setting it up.

Will the final version of this iteration of Cura support double extrusion? I am worried because I've noticed that Ultimaker has no dual extrusion models that I could find, so it might not be of much interest to them in implementing it. The current supported printers all seem to be single-extruder.

Ghostkeeper, 2015-12-25:
The engine still supports multi-extrusion. The new interface could be made to support multi-extrusion via the per-object settings since the extruder of a mesh (and related stuff) is communicated to the engine via a setting. I can't make any promises, but I expect this regression to be mended in due time.

Patola, 2015-12-31:
However the current Cura 15.10alpha interface has double-extrusion-only features like "standby temperature". How can double extrusion be considered merely an enhancement? I disagree with that classification.

Ghostkeeper, 2015-12-31:
Yeah, this repository's issue tracker has only the "enhancement" tag for feature requests as well as enhancements to already existing features.

fieldOfView, 2015-12-31:
@Patola look at it from the bright side: the fact that double-extrusion-only features are starting to appear must mean there is more underway...

Patola, 2015-12-31:
Oh, ok. Sorry then. I must say that I am really liking the nw cura interface and I am eager to test the double extrusion stuff.

Ghostkeeper, 2016-11-28:
Took some work, but this is now implemented.

closed 2016-11-28 by Ghostkeeper