Ultimaker/Cura #580: Feature request: drag and drop import model files

juj, 2016-01-04:
I am running `Cura-1.99.0-dev.20151117-win64.exe` from http://software.ultimaker.com/Cura_closed_beta/15.10/, and I notice that unlike in the previous Cura 15.04.3, it is not currently possible to import model files (.obj / .stl) by dragging a file from Windows Explorer onto the 3D print plate view in the Cura main window. Could it be added in the new version as well?

Ghostkeeper, 2016-01-04:
Oh, that should be possible. It works on Ubuntu, at least. This is how I import models all the time! I'll file this as a bug then. It seems to be Windows-specific then.

juj, 2016-01-05:
Hmm, odd, as testing today, drag and drop does now work for me. I then installed Cura 1.99.0 from the same executable above on a new computer, and there also drag and drop did not work, but only on the first run. Closing the application and reopening, drag and drop started to work there as well.

Ghostkeeper, 2016-01-06:
Gah, I hate those kinds of bugs :stuck_out_tongue:

closed 2016-12-26 by thopiekar