Ultimaker/Cura #581: Confusion about the Global Profile settings

juj, 2016-01-04:
Testing out Cura-1.99.0-dev.20151117-win64.exe from http://software.ultimaker.com/Cura_closed_beta/15.10/ on an Ultimaker 2, and I notice that the Global Profile view in Advanded Print Setup mode has the options Low, High, Normal, Ulti.

In old Cura, when I switched between these options (called Fast, Normal, High, Ulti), the z layer height setting was adjusted as follows:

    fast: 0.15mm
    normal: 0.1mm
    high: 0.06mm
    ulti: 0.04mm

In the Cura 1.99.0 version however, when I switch between the options in the Global Profile dropdown, the Layer Height value in the tab panel below does not change, but stays at `0.1 mm` for all settings. Actually, switching between the different profiles, it looks like all settings (Quality, Shell, Infill, Material, Speed, Travel, etc.) stay the same in the settings panel. Is that the intended behavior or am I misunderstanding how to operate the profiles?

Ghostkeeper, 2016-01-04:
No, that's not intended behaviour. Does an arrow appear next to the layer height setting, by any chance? If you click it, the layer height should revert to the profile's default value.

I'll assume for now that the arrow doesn't appear, in which case this is a bug. The version you're using is quite old and I seem to remember some issues with profile defaults coming around since then, so it might already be fixed. We'll have to test that. No build has been published since then since more secret features are included in our weekly builds since then, which may not be revealed until our next release.

juj, 2016-01-05:
An arrow did not appear. Thanks for the update, it helped me understand how it is supposed to work. I'll check this again once the new release comes out.

BagelOrb, 2017-07-21:
not a problem any more

closed 2017-07-21 by BagelOrb