Ultimaker/Cura #584: Post Processing Plugin scripts support input type different from number like text, string?

danrain, 2016-01-05:
I found that Post Processing Plugin scripts only support input type boolean, number,float . I want to input text ,string but i cant find how to?

[image]

nallath, 2016-01-05:
Using "type": "string" should be possible. 

The mistake I see in your code is that there are two objects with the same name (eg; there are two "a":{} objects).

danrain, 2016-01-05:
you can give me sample code run properly? that code i write for this asking. I did the right but i cant run.

nallath, 2016-01-05:
It should use the same formating as the machine definition stuff found in machines/fdmprinter.json

fieldOfView, 2016-01-06:
Strings may not currently work:
https://github.com/Ultimaker/Uranium/issues/76

nallath, 2016-01-06:
Is that issue also in the Jira tracker?

fieldOfView, 2016-01-06:
It is now (this was from before I had access to jira)

danrain, 2016-01-07:
when have we fixed String issue? I need to write Custom Code Plugin Scripts that having regular expression from input box. Or i must write another private plugin ? Nallath

nallath, 2016-01-07:
We haven't fixed the string issue, but it's logged in our own tracker, so it will be fixed at some point in time.
What is it that you want to do that requires a regex?

danrain, 2016-01-07:
that is post processing plugin script which input a regular expression
string and replace that regex with the existing string inside gcode.

for example :

danrain, 2016-01-07:
[image]

closed 2018-11-22 by Appesteijn