Ultimaker/Cura #585: Oops, uncaught exception on startup.

ze897c, 2016-01-06:
Version: master
Platform: Darwin-15.2.0-x86_64-i386-64bit
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "/Users/mattpatterson/devel/cura-build/build/dist/Cura.app/Contents/Resources/__boot__.py", line 77, in <module>
  File "/Users/mattpatterson/devel/cura-build/build/dist/Cura.app/Contents/Resources/__boot__.py", line 62, in _run
  File "/Users/mattpatterson/devel/cura-build/build/dist/Cura.app/Contents/Resources/cura_app.py", line 16, in <module>
  File "UM/Application.pyc", line 301, in getInstance
  File "cura/CuraApplication.pyc", line 58, in __init__
  File "UM/Qt/QtApplication.pyc", line 78, in __init__
  File "cura/CuraApplication.pyc", line 109, in _loadPlugins
RuntimeError: Could not load the backend plugin!

ze897c, 2016-01-06:
Grrrr...sorry...think I already posted this...sorry

thopiekar, 2016-01-13:
No, there is only this issue report created from you ( https://github.com/Ultimaker/Cura/issues/created_by/ze897c ).
Please add more info about how you installed Cura on your Mac. And more important which version do you use. Your log does no tell anything about it.

thopiekar, 2016-01-13:
Ah, sorry, found the old issue which is already closed: https://github.com/Ultimaker/Cura/issues/253
So the answer is are you now using a more recent version of Cura? Or still the same?

ze897c, 2016-01-14:
Sorry Thomas-Karl, I opened the old version and, no surprise, got the old bug.

I was just being dumb and forgetful.

closed 2016-01-14 by Ghostkeeper