Ultimaker/Cura #590: [2.1] No layers shown after sliding a vase.

thopiekar, 2016-01-14:
I noticed on this vases (http://www.thingiverse.com/thing:37117/#files) and also others on other websites, that vases are very often closed. Is it an viewer issue? The view embedded on the website also shows the model closed.

Additionally Cura is slicing this model very fast and I don't see any error in a console or the cura.log, but when choosing the layer view it shows simply nothing.

Does it also happen to you?

Ghostkeeper, 2016-01-15:
I tried the first of those vases (6MB file). It gives me this error:

    2016-01-15 10:13:02,980 - ERROR - Failed to parse socket error

So that's a bug. I suspect that maybe something changed in either the g-code that CuraEngine returns or in the protobuf definitions.

The reason why you often see vases with closed tops is because they are meant to be sliced with the `spiralize` setting activated. Spiralize makes the surface a bit smoother, and may sometimes help making it more watertight.

BagelOrb, 2016-01-20:
Also the Spiralize feature automatically sets the infill to 0%, the top thickness to 0mm and the bottom thickness 0.1 (same as layer height), so the model is printed as if it is a hollow vase...

thopiekar, 2016-01-20:
Ah, OK, so I see the top closed with no volume!

However, layers are still not generated (no progress bar). Will retry with
newer nightlies soon.

Ghostkeeper, 2016-01-21:
We had an issue with a "socket error" in recent builds, which was due to the profile used by the engine and the profile used by the Python side being out of sync. We fixed that one. There's also this one: #594 which is still open.

thopiekar, 2016-02-07:
Fixed via https://github.com/Ultimaker/Cura/commit/8d2fc1f7e1222f68edaf92f14e02c555641fa0c0

Now Cura is slicing almost everything here and generates layers easily.

closed 2016-02-07 by thopiekar