Ultimaker/Cura #591: [2.1] G-Code filename not set when opening cura with a file

thopiekar, 2016-01-14:
As I own a bq Hephestos 2 the filename for my printer is called BH2_[filename of model]. So when opening a file on Cura it works very well, but when I use a filemanager and open stl file, for example, with Cura, it loads the model, slices it, but doesn't set the filename. So the resulting name is "BH2_".

Ghostkeeper, 2016-01-15:
You're right. I haven't got Cura installed since I always work from source, but I was able to reproduce this issue by calling from the command line on Ubuntu:

    python3 cura_app.py <model file>

The file is loaded but no name is given other than "UM2_".

closed 2016-01-28 by BagelOrb

thopiekar, 2016-01-28:
Can confirm it works now on Linux :+1:

Ghostkeeper, 2016-01-29:
Nice, thanks!

commit a1be5a08 (2016-01-15): Allows for a file to be opened using the terminal

or something alike for instance when the file is dragged onto the logo(MacOS) or with 'open with'(Windows)

Fixes to #CURA-707
Fixes to #CURA-620
Fixes #591