Ultimaker/Cura #593: Feature Request: custom G Code inserted at layer change

hariedo, 2016-01-16:
I am not sure if a plugin modifier can accurately detect the difference between a z-hop and a layer change, so I think it's best if there is a clear and obvious way to insert custom G Code at each layer change, just as you can insert custom G Code at the start and end of a print.

This is the best time to insert things like M240 - trigger a camera, for example.

Ghostkeeper, 2016-01-16:
At every layer change, a comment is inserted that says: `;LAYER: n`. This can be used to accurately detect layer changes, as long as CuraEngine is used as slicer.

Also, the g-code is sent to the front-end per layer, and the front-end stores it as such. The PostProcessing plug-in currently depends on this property in order to be able to insert things between layers without having to parse and split the g-code.

The PostProcessing plug-in (https://github.com/nallath/PostProcessingPlugin) is planned to be merged into Cura eventually. A script could be added to insert arbitrary g-code after every layer.

nallath, 2016-01-18:
The PostProcessing plugin is an offical part of Cura (it is in the current release).

closed 2018-11-22 by Appesteijn