Ultimaker/Cura #594: 2.099-2.1: Stuck "Preparing to slice"

CCS86, 2016-01-16:
x64 on Win10

On my first test drive of this version, I loaded a simple model, and found Cura sticks right here. It happens with any model so far:

[image]

Ghostkeeper, 2016-01-16:
Does the normal log say something in `C:\Users\<you>\AppData\Local\cura`?

We tested this build on Win10 but didn't have this problem.

CCS86, 2016-01-16:
Looks like it could be engine related:

2016-01-15 23:03:33,469 - WARNING - Could not determine application version from string 2.0.99-2.1, not checking for updates
2016-01-15 23:03:34,499 - DEBUG - Loaded theme C:\Program Files\Cura 2.0\resources\themes\cura
2016-01-15 23:03:41,312 - DEBUG - Initialized OpenGL subsystems.
2016-01-15 23:03:41,313 - DEBUG - OpenGL Version:  4.5.0 NVIDIA 358.91
2016-01-15 23:03:41,313 - DEBUG - OpenGL Vendor:   NVIDIA Corporation
2016-01-15 23:03:41,313 - DEBUG - OpenGL Renderer: GeForce GTX 970M/PCIe/SSE2
2016-01-15 23:04:41,529 - DEBUG - Autosaving preferences, instances and profiles
2016-01-15 23:05:41,531 - DEBUG - Autosaving preferences, instances and profiles
2016-01-15 23:17:13,752 - DEBUG - Calculating normals took 0.0030019283294677734 seconds
2016-01-15 23:17:13,753 - DEBUG - Loaded a mesh with 69048 vertices
2016-01-15 23:17:13,808 - DEBUG - Loading mesh took 2.4514849185943604 seconds
2016-01-15 23:17:24,226 - DEBUG - Setting active view to LayerView
2016-01-15 23:18:14,013 - DEBUG - Autosaving preferences, instances and profiles
2016-01-15 23:20:06,472 - DEBUG - Setting active view to SolidView
2016-01-15 23:20:08,563 - DEBUG - Sending data to engine for slicing.
2016-01-15 23:20:19,555 - DEBUG - Setting value of setting line_width to 0.3
2016-01-15 23:20:22,691 - DEBUG - Setting value of setting line_width to 0.4
2016-01-15 23:20:23,450 - DEBUG - Sending data to engine for slicing.
2016-01-15 23:20:26,877 - DEBUG - Setting value of setting wall_line_count to 3
2016-01-15 23:20:43,252 - DEBUG - Setting value of setting fill_perimeter_gaps to everywhere
2016-01-15 23:20:43,846 - DEBUG - Sending data to engine for slicing.
2016-01-15 23:20:54,658 - DEBUG - Setting value of setting infill_sparse_density to 1
2016-01-15 23:20:54,870 - DEBUG - Setting value of setting infill_sparse_density to 10
2016-01-15 23:20:55,011 - DEBUG - Setting value of setting infill_sparse_density to 100
2016-01-15 23:21:04,565 - DEBUG - Sending data to engine for slicing.
2016-01-15 23:21:04,666 - DEBUG - Setting value of setting infill_overlap to 1
2016-01-15 23:22:07,063 - DEBUG - Setting value of setting coasting_volume to 0.2
2016-01-15 23:22:07,626 - DEBUG - Sending data to engine for slicing.
2016-01-15 23:22:13,202 - DEBUG - Setting value of setting coasting_min_volume to 2
2016-01-15 23:22:15,584 - DEBUG - Setting value of setting coasting_speed to 1
2016-01-15 23:22:15,793 - DEBUG - Setting value of setting coasting_speed to 10
2016-01-15 23:22:15,915 - DEBUG - Setting value of setting coasting_speed to 100
2016-01-15 23:22:16,488 - DEBUG - Sending data to engine for slicing.
2016-01-15 23:22:44,530 - DEBUG - Setting value of setting layer_height_0 to 0.1
2016-01-15 23:23:23,605 - DEBUG - Sending data to engine for slicing.
2016-01-15 23:24:14,542 - DEBUG - Autosaving preferences, instances and profiles
2016-01-15 23:25:14,552 - DEBUG - Autosaving preferences, instances and profiles
2016-01-15 23:25:38,439 - DEBUG - Calculating normals took 0.0 seconds
2016-01-15 23:25:38,439 - DEBUG - Loaded a mesh with 14160 vertices
2016-01-15 23:25:38,469 - DEBUG - Loading mesh took 0.6885569095611572 seconds
2016-01-15 23:25:39,201 - DEBUG - Sending data to engine for slicing.
2016-01-15 23:26:38,672 - DEBUG - Autosaving preferences, instances and profiles

CCS86, 2016-01-16:
This log is from a fresh application load (this time as administrator), and loading my model. I don't see it even trying to send data for slicing:

2016-01-16 13:30:33,816 - INFO - Loaded plugin FileLogger
2016-01-16 13:30:33,827 - INFO - Loaded plugin GCodeProfileReader
2016-01-16 13:30:33,839 - INFO - Loaded plugin GCodeWriter
2016-01-16 13:30:33,853 - INFO - Loaded plugin ImageReader
2016-01-16 13:30:33,868 - INFO - Loaded plugin LayerView
2016-01-16 13:30:33,882 - INFO - Loaded plugin LegacyProfileReader
2016-01-16 13:30:33,898 - INFO - Loaded plugin LocalFileOutputDevice
2016-01-16 13:30:33,916 - INFO - Loaded plugin PerObjectSettingsTool
2016-01-16 13:30:33,935 - INFO - Loaded plugin PostProcessingPlugin
2016-01-16 13:30:33,958 - INFO - Loaded plugin RemovableDriveOutputDevice
2016-01-16 13:30:34,018 - INFO - Loaded plugin SliceInfoPlugin
2016-01-16 13:30:34,036 - INFO - Loaded plugin SolidView
2016-01-16 13:30:34,054 - INFO - Loaded plugin CameraTool
2016-01-16 13:30:34,077 - INFO - Loaded plugin MirrorTool
2016-01-16 13:30:34,102 - INFO - Loaded plugin RotateTool
2016-01-16 13:30:34,127 - INFO - Loaded plugin ScaleTool
2016-01-16 13:30:34,154 - INFO - Loaded plugin SelectionTool
2016-01-16 13:30:34,177 - INFO - Loaded plugin TranslateTool
2016-01-16 13:30:34,201 - INFO - Checking for new version of cura
2016-01-16 13:30:34,201 - INFO - Loaded plugin UpdateChecker
2016-01-16 13:30:34,242 - INFO - Loaded plugin USBPrinting
2016-01-16 13:30:34,266 - INFO - Loaded plugin SimpleView
2016-01-16 13:30:34,291 - INFO - Loaded plugin WireframeView
2016-01-16 13:30:34,319 - INFO - Loaded plugin XRayView
2016-01-16 13:30:34,383 - ERROR - Invalid type for Setting machine_disallowed_areas, ignoring
2016-01-16 13:30:34,438 - DEBUG - Setting value of setting material_bed_temperature to 70
2016-01-16 13:30:34,438 - DEBUG - Setting value of setting adhesion_type to brim
2016-01-16 13:30:34,438 - DEBUG - Setting value of setting coasting_volume to 0.2
2016-01-16 13:30:34,439 - DEBUG - Setting value of setting support_infill_rate to 13
2016-01-16 13:30:34,439 - DEBUG - Setting value of setting speed_infill to 35
2016-01-16 13:30:34,439 - DEBUG - Setting value of setting speed_print to 30
2016-01-16 13:30:34,439 - DEBUG - Setting value of setting layer_height to 0.10
2016-01-16 13:30:34,439 - DEBUG - Setting value of setting infill_wipe_dist to 0.3
2016-01-16 13:30:34,440 - DEBUG - Setting value of setting layer_height_0 to 0.1
2016-01-16 13:30:34,440 - DEBUG - Setting value of setting support_xy_distance to 1.2
2016-01-16 13:30:34,440 - DEBUG - Setting value of setting skirt_line_count to 3
2016-01-16 13:30:34,440 - DEBUG - Setting value of setting support_angle to 80
2016-01-16 13:30:34,441 - DEBUG - Setting value of setting support_join_distance to 3
2016-01-16 13:30:34,441 - DEBUG - Setting value of setting cool_lift_head to True
2016-01-16 13:30:34,441 - DEBUG - Setting value of setting retraction_amount to 3.5
2016-01-16 13:30:34,441 - DEBUG - Setting value of setting retraction_retract_speed to 24.0
2016-01-16 13:30:34,442 - DEBUG - Setting value of setting support_bottom_stair_step_height to 0.2
2016-01-16 13:30:34,442 - DEBUG - Setting value of setting retraction_extrusion_window to 4
2016-01-16 13:30:34,442 - DEBUG - Setting value of setting support_type to buildplate
2016-01-16 13:30:34,442 - DEBUG - Setting value of setting top_layers to 4
2016-01-16 13:30:34,442 - DEBUG - Setting value of setting support_roof_height to 0.3
2016-01-16 13:30:34,443 - DEBUG - Setting value of setting infill_sparse_density to 100
2016-01-16 13:30:34,443 - DEBUG - Setting value of setting retraction_prime_speed to 18
2016-01-16 13:30:34,443 - DEBUG - Setting value of setting material_print_temperature to 190
2016-01-16 13:30:34,443 - DEBUG - Setting value of setting coasting_speed to 100
2016-01-16 13:30:34,443 - DEBUG - Setting value of setting coasting_enable to True
2016-01-16 13:30:34,444 - DEBUG - Setting value of setting support_conical_min_width to 5
2016-01-16 13:30:34,444 - DEBUG - Setting value of setting wall_line_count to 3
2016-01-16 13:30:34,444 - DEBUG - Setting value of setting speed_layer_0 to 15.0
2016-01-16 13:30:34,444 - DEBUG - Setting value of setting coasting_min_volume to 2
2016-01-16 13:30:34,444 - DEBUG - Setting value of setting support_enable to False
2016-01-16 13:30:34,445 - DEBUG - Setting value of setting bottom_layers to 4
2016-01-16 13:30:34,445 - DEBUG - Setting value of setting travel_compensate_overlapping_walls_enabled to True
2016-01-16 13:30:34,445 - DEBUG - Setting value of setting retraction_count_max to 5
2016-01-16 13:30:34,445 - DEBUG - Setting value of setting infill_overlap to 1
2016-01-16 13:30:34,445 - DEBUG - Setting value of setting speed_travel to 200
2016-01-16 13:30:34,445 - DEBUG - Setting value of setting support_offset to .4
2016-01-16 13:30:34,446 - DEBUG - Setting value of setting retraction_hop to 0.05
2016-01-16 13:30:34,446 - DEBUG - Setting value of setting support_roof_enable to True
2016-01-16 13:30:34,446 - DEBUG - Setting value of setting support_roof_pattern to lines
2016-01-16 13:30:34,446 - DEBUG - Setting value of setting z_seam_type to back
2016-01-16 13:30:34,447 - DEBUG - Setting value of setting support_use_towers to False
2016-01-16 13:30:34,447 - DEBUG - Setting value of setting brim_line_count to 5
[... 41 lines cut ...]
2016-01-16 13:30:34,465 - DEBUG - Setting value of setting shell_thickness to 0.8
[... cut ...]
2016-01-16 13:30:41,117 - DEBUG - Initialized OpenGL subsystems.
2016-01-16 13:30:41,118 - DEBUG - OpenGL Version:  4.5.0 NVIDIA 358.91
2016-01-16 13:30:41,118 - DEBUG - OpenGL Vendor:   NVIDIA Corporation
2016-01-16 13:30:41,118 - DEBUG - OpenGL Renderer: GeForce GTX 970M/PCIe/SSE2
2016-01-16 13:30:54,075 - DEBUG - Calculating normals took 0.0035037994384765625 seconds
2016-01-16 13:30:54,075 - DEBUG - Loaded a mesh with 69048 vertices
2016-01-16 13:30:54,165 - DEBUG - Loading mesh took 0.4911930561065674 seconds

CCS86, 2016-01-16:
If I drag my model somewhere else on the platform, these two lines are added:

2016-01-16 13:31:54,391 - DEBUG - Autosaving preferences, instances and profiles
2016-01-16 13:32:44,218 - DEBUG - Sending data to engine for slicing.

CCS86, 2016-01-20:
Any thoughts?

nallath, 2016-01-20:
I've encountered this as well. Are you printing one at a time?

bburrough, 2016-01-20:
I have this problem, too, and I'm not printing one at a time.

bburrough, 2016-01-20:
For reference: CURA-723

CCS86, 2016-01-21:
I am set to all at once. Single object.

CCS86, 2016-01-28:
Any ideas?

Ghostkeeper, 2016-01-29:
Yes, we had an incongruence in the protocol between front-end and back-end when I gave you that build. It has since been amended, sort of, then broken again, and now @ahiemstra is working on a more permanent fix and I'm adjusting the protocol again too. I am allowed to supply @CCS86 and @bburrough with new builds but I'd like to wait for our two issues to be done. They are marked CURA-434 and CURA-427 in our internal issue tracker, Bob.

CCS86, 2016-01-29:
Sounds good, keep us posted!

CCS86, 2016-02-06:
Are we getting close to a new beta build?

nallath, 2016-02-08:
Yep, so close!

Ghostkeeper, 2016-02-08:
But this issue is still stubbornly clinging on.

CCS86, 2016-02-15:
Good to hear, I'm itching to try it out! :)

CCS86, 2016-04-05:
This happened again for me in the 3/18 build. Everything was slicing fine until I enabled support (pretty sure that was it). It got stuck preparing to slice and I can't seem to get it fired up again. I've tried disabling support, moving the object, reloading the object, changing other settings.

Here is the log data:

`2016-04-04 20:42:22,501 - INFO - Loaded plugin FileLogger
2016-04-04 20:42:22,515 - INFO - Loaded plugin GCodeProfileReader
2016-04-04 20:42:22,528 - INFO - Loaded plugin GCodeWriter
2016-04-04 20:42:22,545 - INFO - Loaded plugin ImageReader
2016-04-04 20:42:22,563 - INFO - Loaded plugin LayerView
2016-04-04 20:42:22,578 - INFO - Loaded plugin LegacyProfileReader
2016-04-04 20:42:22,594 - INFO - Loaded plugin LocalFileOutputDevice
2016-04-04 20:42:22,614 - INFO - Loaded plugin PerObjectSettingsTool
2016-04-04 20:42:22,636 - INFO - Loaded plugin PostProcessingPlugin
2016-04-04 20:42:22,660 - INFO - Loaded plugin RemovableDriveOutputDevice
2016-04-04 20:42:22,685 - INFO - Loaded plugin SliceInfoPlugin
2016-04-04 20:42:22,703 - INFO - Loaded plugin SolidView
2016-04-04 20:42:22,722 - INFO - Loaded plugin CameraTool
2016-04-04 20:42:22,748 - INFO - Loaded plugin MirrorTool
2016-04-04 20:42:22,777 - INFO - Loaded plugin RotateTool
2016-04-04 20:42:22,802 - INFO - Loaded plugin ScaleTool
2016-04-04 20:42:22,825 - DEBUG - Connected Boxes: {}
2016-04-04 20:42:22,826 - INFO - Loaded plugin SelectionTool
2016-04-04 20:42:22,850 - INFO - Loaded plugin TranslateTool
2016-04-04 20:42:22,874 - INFO - Checking for new version of cura
2016-04-04 20:42:22,875 - INFO - Loaded plugin UpdateChecker
2016-04-04 20:42:22,917 - INFO - Loaded plugin USBPrinting
2016-04-04 20:42:22,942 - INFO - Loaded plugin SimpleView
2016-04-04 20:42:22,966 - INFO - Loaded plugin WireframeView
2016-04-04 20:42:22,993 - INFO - Loaded plugin XRayView
2016-04-04 20:42:22,994 - INFO - Command line arguments: {'debug-mode': False, 'file': [], 'external-backend': False, 'disable-textures': False}
2016-04-04 20:42:23,156 - WARNING - No active machine found when trying to save setting visibility
2016-04-04 20:42:23,160 - DEBUG - Attempting to kill the engine process
2016-04-04 20:42:23,419 - WARNING - Could not determine application version from string 2.1.0-20160318, not checking for updates
2016-04-04 20:42:23,449 - INFO - Started engine process: C:\Program Files\Cura 2.1_3-18\CuraEngine.exe
2016-04-04 20:42:23,483 - DEBUG - Setting active view to SolidView
2016-04-04 20:42:23,636 - DEBUG - Calculating normals took 0.0004839897155761719 seconds
2016-04-04 20:42:23,636 - DEBUG - Loaded a mesh with 10452 vertices
2016-04-04 20:42:24,404 - DEBUG - Loaded theme C:\Program Files\Cura 2.1_3-18\resources\themes\cura
2016-04-04 20:42:25,104 - DEBUG - Backend connected on port 49674
2016-04-04 20:42:25,412 - DEBUG - Initialized OpenGL subsystems.
2016-04-04 20:42:25,413 - DEBUG - OpenGL Version:  4.5.0 NVIDIA 364.72
2016-04-04 20:42:25,413 - DEBUG - OpenGL Vendor:   NVIDIA Corporation
2016-04-04 20:42:25,413 - DEBUG - OpenGL Renderer: GeForce GTX 970M/PCIe/SSE2
2016-04-04 20:42:28,143 - DEBUG - Connected Boxes: {}
2016-04-04 20:42:33,474 - DEBUG - Connected Boxes: {}
2016-04-04 20:42:35,616 - DEBUG - Autosaving preferences, instances and profiles
2016-04-04 20:42:38,789 - DEBUG - Connected Boxes: {}
2016-04-04 20:42:44,110 - DEBUG - Connected Boxes: {}
2016-04-04 20:42:49,418 - DEBUG - Connected Boxes: {}
2016-04-04 20:42:54,732 - DEBUG - Connected Boxes: {}
2016-04-04 20:43:00,041 - DEBUG - Connected Boxes: {}
2016-04-04 20:43:05,356 - DEBUG - Connected Boxes: {}
2016-04-04 20:43:10,670 - DEBUG - Connected Boxes: {}
2016-04-04 20:43:15,989 - DEBUG - Connected Boxes: {}
2016-04-04 20:43:21,310 - DEBUG - Connected Boxes: {}
2016-04-04 20:43:26,619 - DEBUG - Connected Boxes: {}
2016-04-04 20:43:32,048 - DEBUG - Connected Boxes: {}
2016-04-04 20:43:37,375 - DEBUG - Connected Boxes: {}
2016-04-04 20:43:39,810 - DEBUG - Calculating normals took 0.002496957778930664 seconds
2016-04-04 20:43:39,811 - DEBUG - Loaded a mesh with 95826 vertices
2016-04-04 20:43:39,862 - DEBUG - Loading mesh took 1.6668860912322998 seconds
2016-04-04 20:43:40,562 - DEBUG - Attempting to kill the engine process
2016-04-04 20:43:40,562 - DEBUG - Killing engine process
2016-04-04 20:43:40,563 - DEBUG - Engine process is killed
2016-04-04 20:43:40,565 - INFO - Backend crashed or closed. Restarting...
2016-04-04 20:43:40,755 - INFO - Started engine process: C:\Program Files\Cura 2.1_3-18\CuraEngine.exe
2016-04-04 20:43:40,813 - DEBUG - Backend connected on port 49674
2016-04-04 20:43:41,327 - DEBUG - Sending data to engine for slicing.
2016-04-04 20:43:41,327 - DEBUG - Sending data to engine is completed
2016-04-04 20:43:42,696 - DEBUG - Connected Boxes: {}
[... 208 lines cut ...]
2016-04-04 20:49:20,835 - DEBUG - Autosaving preferences, instances and profiles
2016-04-04 20:49:24,995 - DEBUG - Connected Boxes: {}
2016-04-04 20:49:30,328 - DEBUG - Connected Boxes: {}
2016-04-04 20:49:33,235 - DEBUG - Attempting to kill the engine process
2016-04-04 20:49:34,128 - DEBUG - Sending data to engine for slicing.
2016-04-04 20:49:34,129 - DEBUG - Sending data to engine is completed
2016-04-04 20:49:35,635 - DEBUG - Connected Boxes: {}
2016-04-04 20:49:37,621 - DEBUG - Calculating normals took 0.0024509429931640625 seconds
2016-04-04 20:49:37,621 - DEBUG - Loaded a mesh with 95826 vertices
2016-04-04 20:49:37,671 - DEBUG - Loading mesh took 1.1319379806518555 seconds
[... cut ...]
2016-04-04 20:49:53,138 - DEBUG - Sending data to engine for slicing.
2016-04-04 20:49:53,138 - DEBUG - Sending data to engine is completed
2016-04-04 20:49:53,413 - DEBUG - Attempting to kill the engine process
2016-04-04 20:49:54,107 - DEBUG - Sending data to engine for slicing.
2016-04-04 20:49:54,107 - DEBUG - Sending data to engine is completed
2016-04-04 20:49:57,022 - DEBUG - Connected Boxes: {}
2016-04-04 20:50:02,337 - DEBUG - Connected Boxes: {}
2016-04-04 20:50:07,676 - DEBUG - Connected Boxes: {}
2016-04-04 20:50:13,004 - DEBUG - Connected Boxes: {}`

nallath, 2016-04-05:
Yeah, this is a known issue. I'm pretty sure I fixed this, but it's not in the 3/18 build yet.

Ghostkeeper, 2016-04-05:
The support and brim dependency loops have been fixed around the 28th of March or so. With the current plan, we're entering open beta on Friday and there'll be new builds to be made public then.

g3gg0, 2016-04-14:
i would like to re-open this issue.

i am using Cura 2.1.0 and when moving a part (star symbol) with 2mm height and ~8x8cm footprint,
the GUI hangs with error message "Preparing to slice...".
To reproduce - just place a part, simple settings, ulti quality,  and repeatedly move it on the build plate.

When it hangs, the process CuraEngine.exe is non-existent, manually connecting fails:

```
C:\Users\edf76fg>"C:\Program Files\Cura 2.1\CuraEngine.exe" connect 127.0.0.1:49674 -j "C:\Program Files\Cura 2.1\resources\machines\ultimaker2.json" -vv

Cura_SteamEngine version DEV
[license]

Loading polygon setting machine_disallowed_areas not implemented...
Unrecognized data type in JSON: machine_platform_offset has type array
Loading polygon setting machine_head_polygon not implemented...
Loading polygon setting machine_head_with_fans_polygon not implemented...
Unrecognized data type in JSON: machine_head_with_fans_polygon has type array
Unrecognized data type in JSON: machine_disallowed_areas has type array
Unrecognized data type in JSON: machine_platform_offset has type array
Connecting to 127.0.0.1:49674
Arcus Fatal Error (2, native 10061): Could not connect to the given address
Connected to 127.0.0.1:49674
Closing connection
```

The log file of cura which shows that the last few connects seemed to have failed

```

2016-04-14 12:24:52,405 - DEBUG - Engine process is killed. Recieved return code 1
2016-04-14 12:24:52,418 - INFO - Backend crashed or closed. Restarting...
2016-04-14 12:24:52,522 - DEBUG - Attempting to kill the engine process
2016-04-14 12:24:52,558 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
2016-04-14 12:24:52,735 - DEBUG - Backend connected on port 49674
2016-04-14 12:24:53,261 - DEBUG - Sending data to engine for slicing.
2016-04-14 12:24:53,261 - DEBUG - Sending data to engine is completed
2016-04-14 12:24:53,285 - DEBUG - Attempting to kill the engine process
2016-04-14 12:24:53,285 - DEBUG - Killing engine process
2016-04-14 12:24:53,288 - DEBUG - Engine process is killed. Recieved return code 1
2016-04-14 12:24:53,303 - INFO - Backend crashed or closed. Restarting...
2016-04-14 12:24:53,408 - DEBUG - Attempting to kill the engine process
2016-04-14 12:24:53,457 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
2016-04-14 12:24:53,654 - DEBUG - Backend connected on port 49674
2016-04-14 12:24:54,204 - DEBUG - Sending data to engine for slicing.
2016-04-14 12:24:54,204 - DEBUG - Sending data to engine is completed
2016-04-14 12:24:54,853 - DEBUG - Attempting to kill the engine process
2016-04-14 12:24:54,853 - DEBUG - Killing engine process
2016-04-14 12:24:54,857 - DEBUG - Engine process is killed. Recieved return code 1
2016-04-14 12:24:54,894 - INFO - Backend crashed or closed. Restarting...
2016-04-14 12:24:54,999 - DEBUG - Attempting to kill the engine process
2016-04-14 12:24:55,047 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
2016-04-14 12:24:55,233 - DEBUG - Backend connected on port 49674
2016-04-14 12:24:55,775 - DEBUG - Sending data to engine for slicing.
2016-04-14 12:24:55,775 - DEBUG - Sending data to engine is completed
2016-04-14 12:24:57,620 - DEBUG - Backend quit with return code 0. Resetting process and socket.
2016-04-14 12:24:57,633 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
2016-04-14 12:24:57,815 - DEBUG - Backend connected on port 49674
2016-04-14 12:24:58,607 - DEBUG - Attempting to kill the engine process
2016-04-14 12:24:58,607 - DEBUG - Killing engine process
2016-04-14 12:24:58,613 - DEBUG - Engine process is killed. Recieved return code 1
2016-04-14 12:24:59,063 - INFO - Backend crashed or closed. Restarting...
2016-04-14 12:24:59,167 - DEBUG - Attempting to kill the engine process
2016-04-14 12:24:59,170 - INFO - Started engine process: C:\Program Files\Cura 2.1\CuraEngine.exe
2016-04-14 12:24:59,307 - DEBUG - Sending data to engine for slicing.
2016-04-14 12:24:59,308 - DEBUG - Sending data to engine is completed
2016-04-14 12:24:59,332 - DEBUG - Attempting to kill the engine process
2016-04-14 12:24:59,332 - DEBUG - Killing engine process
2016-04-14 12:24:59,334 - DEBUG - Engine process is killed. Recieved return code 1
2016-04-14 12:24:59,928 - DEBUG - Sending data to engine for slicing.
2016-04-14 12:24:59,928 - DEBUG - Sending data to engine is completed
2016-04-14 12:25:00,707 - DEBUG - Attempting to kill the engine process
2016-04-14 12:25:01,407 - DEBUG - Sending data to engine for slicing.
2016-04-14 12:25:01,407 - DEBUG - Sending data to engine is completed
2016-04-14 12:25:03,608 - DEBUG - Attempting to kill the engine process
2016-04-14 12:25:04,230 - DEBUG - Attempting to kill the engine process
2016-04-14 12:25:04,810 - DEBUG - Attempting to kill the engine process
2016-04-14 12:25:05,494 - DEBUG - Sending data to engine for slicing.
2016-04-14 12:25:05,494 - DEBUG - Sending data to engine is completed
```

sedwards2009, 2016-04-14:
I'm getting this problem on master right now. (Win 10, 64bit)

nilrog, 2016-04-15:
I got this issue too with the current open beta.

1. Loaded three small objects by dropping them all on Cura at the same time
2. Changed layer height and infill
3. Changed print sequence to "one at a time"
4. Had to manually reposition the objects because Cura place one of them outside of the print area (another bug imo)
5. Fine tuned the positions a bit more
6. Cura stopped slicing...it is just showing "Preparing to slice". It doesn't help to modify anything...it will not slice again.

g3gg0, 2016-04-18:
hi.
is any information missing / can the devs reproduce this error?

nallath, 2016-04-18:
sedwards2009 is part of the team, so yes

g3gg0, 2016-04-18:
perfect, thanks

Gareee, 2016-04-18:
Yeah I can confirm reproducible and also windows 10 64 bit as well.

fieldOfView, 2016-04-19:
Try as I might, I cannot seem to reproduce this on Windows 7 64 bit. Anyone on Win7 with this bug, or could this be a Windows 10-only bug?

nilrog, 2016-04-19:
Yes, look a few comments above.

https://github.com/Ultimaker/Cura/issues/594#issuecomment-210305078

fieldOfView, 2016-04-19:
@nilrog: doh! I read right over that. Thanks.

g3gg0, 2016-04-28:
did anyone find a workaround for this issue?
tried various things like stalling processes until i finished movement then continuing it etc.
but i cannot find a way to print multiple parts at once :(

Ghostkeeper, 2016-04-29:
As far as I've heard, some work was done in an effort to fix this in the past week, but this caused a different bug to surface which causes the slicing to restart in some cases just as it's completed.

g3gg0, 2016-04-29:
oh, sad to hear. thanks for the update!

CCS86, 2016-05-27:
On 2.1.1, I have run into this.

It seemed to crop up after doing a "lay flat".

Log data:

`2016-05-27 10:50:24,307 - DEBUG - Calculating normals took 0.0005161762237548828 seconds
2016-05-27 10:50:24,307 - DEBUG - Loaded a mesh with 14394 vertices
2016-05-27 10:50:24,312 - DEBUG - Loading mesh took 0.13719701766967773 seconds
2016-05-27 10:50:26,268 - DEBUG - Backend quit with return code 0. Resetting process and socket.
2016-05-27 10:50:26,407 - INFO - Started engine process: C:\Program Files\Cura 2.1.1 open\CuraEngine.exe
2016-05-27 10:50:26,478 - DEBUG - Backend connected on port 49674
2016-05-27 10:50:34,514 - DEBUG - Autosaving preferences, instances and profiles
2016-05-27 10:50:45,589 - DEBUG - Attempting to kill the engine process
2016-05-27 10:50:45,589 - DEBUG - Killing engine process
2016-05-27 10:50:45,591 - DEBUG - Engine process is killed. Recieved return code 1
2016-05-27 10:50:45,992 - INFO - Backend crashed or closed. Restarting...
2016-05-27 10:50:46,095 - DEBUG - Attempting to kill the engine process
2016-05-27 10:50:46,215 - INFO - Started engine process: C:\Program Files\Cura 2.1.1 open\CuraEngine.exe
2016-05-27 10:50:46,269 - DEBUG - Backend connected on port 49674
2016-05-27 10:50:46,764 - DEBUG - Attempting to kill the engine process
2016-05-27 10:50:46,764 - DEBUG - Killing engine process
2016-05-27 10:50:46,766 - DEBUG - Engine process is killed. Recieved return code 1
2016-05-27 10:50:46,771 - INFO - Backend crashed or closed. Restarting...
2016-05-27 10:50:46,874 - DEBUG - Attempting to kill the engine process
2016-05-27 10:50:46,986 - INFO - Started engine process: C:\Program Files\Cura 2.1.1 open\CuraEngine.exe
2016-05-27 10:50:47,041 - DEBUG - Backend connected on port 49674
2016-05-27 10:50:48,847 - DEBUG - Backend quit with return code 0. Resetting process and socket.
2016-05-27 10:50:48,967 - INFO - Started engine process: C:\Program Files\Cura 2.1.1 open\CuraEngine.exe
2016-05-27 10:50:49,023 - DEBUG - Backend connected on port 49674
2016-05-27 10:50:53,355 - DEBUG - Attempting to kill the engine process
2016-05-27 10:50:53,355 - DEBUG - Killing engine process
2016-05-27 10:50:53,358 - DEBUG - Engine process is killed. Recieved return code 1
2016-05-27 10:50:53,548 - INFO - Backend crashed or closed. Restarting...
2016-05-27 10:50:53,652 - DEBUG - Attempting to kill the engine process
2016-05-27 10:50:53,784 - INFO - Started engine process: C:\Program Files\Cura 2.1.1 open\CuraEngine.exe
2016-05-27 10:50:53,901 - DEBUG - Backend connected on port 49674
2016-05-27 10:50:56,731 - DEBUG - Attempting to kill the engine process
2016-05-27 10:50:56,731 - DEBUG - Killing engine process
2016-05-27 10:50:56,731 - DEBUG - Engine process is killed. Recieved return code 0
2016-05-27 10:51:15,408 - DEBUG - Attempting to kill the engine process
2016-05-27 10:51:38,322 - DEBUG - Calculating normals took 0.0004551410675048828 seconds
2016-05-27 10:51:38,322 - DEBUG - Loaded a mesh with 14394 vertices
2016-05-27 10:51:38,327 - DEBUG - Loading mesh took 0.1380770206451416 seconds
2016-05-27 10:51:39,030 - DEBUG - Attempting to kill the engine process
2016-05-27 10:51:41,779 - DEBUG - Attempting to kill the engine process
2016-05-27 10:51:56,030 - DEBUG - Autosaving preferences, instances and profiles`

TheNuclearCat, 2016-08-20:
Same problem :/

kr15uk, 2016-08-25:
Same.
OSX 10.11, all up to date.
Was having this issue on 2.1.2 every 10th or so print.
When seen 2.1.3 realised I was hoping this thing will be fixed, but nah...
Quite annoying, before I had simple fix, quit and reopen but now seems to be it doesn't work.
I'd suspect dodgy STL but this is calibration cube of thingiverse which I slice and print often (with every new filament 3-4 cubes are getting printed to test settings etc)
I've tried to do netfab's repair just to be safe, but nope...

I'm getting more convinced that Simplify3D is probably a good investment... :(

nallath, 2016-08-26:
This is fixed for the next released. 

It also has nothing to do with the quality of your stl.

closed 2016-08-26 by nallath