Ultimaker/Cura #595: [2.1] bq Hephestos 2 build plate is reversed

thopiekar, 2016-01-16:
While working now with Cura and browsing the layers from different positions I always orientate on the two screws in the front of the build plate. When I started the program on my machine I noticed my model got printed on the other side (y-axis). The reason is that the STL file for the build plate is currently wrong and needs to be rotated at the z-axis by 180°.
Tried to do this via MeshLab, but I wasn't successful.

Maybe @nicanor-romero from bq, who uploaded the stl, can fix that quickly :)

thopiekar, 2016-01-20:
Contacted the bq support and here are the requested screenshots and other info they wanted:

In the screenshot above you see Cura with Hephestos 2 loaded as targeted printer. You can see the position of the screw hovered, which is usually at the printers back.

[image]

See:

[image]

So as you orientate on the screws and you think you place your object in the front of the plate Cura will (of course and correctly) generate code for the object at the back :smile: 

Here is the STL file, which is hosted in the repository:
https://github.com/Ultimaker/Cura/blob/master/resources/meshes/bq_hephestos_2_platform.stl
And the commit by bq:
https://github.com/Ultimaker/Cura/commit/bcd863d33b99c6dab201343fe5a232a213412c21

Ghostkeeper, 2016-01-21:
Done: 29a848d90c508c2799dc5b8311c9ac2490502bdc

Ghostkeeper, 2016-01-21:
Thanks, the screenshots helped a lot :smile:

thopiekar, 2016-01-22:
Many thanks! I just tested it now and it seems that you were getting the same problem like me in MeshLab. After rotation the plate has moved in x direction:
[image]

Ghostkeeper, 2016-01-22:
Ah, I'm guessing it wasn't completely centred around the Z axis. I'll see if I can move it a bit around to something that looks good.

thopiekar, 2016-01-22:
Will give feedback on the new commit when I have new builds here.
If the platform is aligned correctly I'll close this of course :wink:

Ghostkeeper, 2016-01-22:
I shifted the platform by 12 millimetres. Looks about centred on my computer.

thopiekar, 2016-01-24:
Worked and the result looks great! Thanks again! :+1:

closed 2016-01-24 by thopiekar