Ultimaker/Cura #596: [2.1] moving all progressbars to a general place

thopiekar, 2016-01-16:
I often wonder about, why the progressbar for slicing is in the left bar and the layer generation progressbar is shown in a popup window.
Wouldn't it be better so show every progress in such a popup window?

Ghostkeeper, 2016-01-16:
We experimented with that, but found that it draws too much attention to put the slicing progress in a pop-up window, with the auto-slicing and all. That's why the slicing is down on the right where it doesn't draw too much attention, but the rest of the progress bars (which do require some attention) are still in a pop-up.

thopiekar, 2016-01-17:
I personally find it confusing.. One popup per object would be good.
Generating layers depends on a finished slicing task, so wouldn't it be great to have only one popup saying: "Preparing object [objectname]: [task]", with the total progress? Additionally it would save space on the screen.
However, just an idea :smiley:

Ghostkeeper, 2016-01-18:
Not everyone is generating layers though. That happens when you switch to layer view (or indeed if you are already in layer view).

fieldOfView, 2016-01-18:
@thopiekar: I was of the same opinion (see here for the same discussion: https://github.com/Ultimaker/Cura/issues/194#issuecomment-127009794). For a short while, we had it like that; all progress in a single place (see the 15.10 betas). This was found to be too jarring though; a progressbar kept popping up (and disappearing) while (auto-)slicing.

In the end we decided on dividing processes in two classes: processes that are blocking to the workflow (though not necessarily to the application), such as loading an object, and processes that happen in the background and allow the user to continue using Cura, such as slicing or printing. The former progress is shown in the Message Stack, the latter in the Sidebar.

I agree that while working in the layers viewmode it is currently a bit strange. When in layers viewmode, slicing becomes a workflow-blocking process, so we should see about showing the slicing progress and loading of the layer stack in a single message.

nallath, 2017-07-27:
Closing as this is such an old issue and the UI already saw a number of changes since then.

closed 2017-07-27 by nallath