Ultimaker/Cura #598: [2.1] KDE: Add suspend inhibition when printing via USB

thopiekar, 2016-01-16:
I think this will be very useful when using a laptop or even a desktop computer, which uses suspending a lot. I found this feature in Amarok together with the KDE desktop. The usage there is to inhibit suspending the system as long as playing music.
So in our case this would be useful during USB printing!

I found the commit which adds the initial feature to Amarok:
https://quickgit.kde.org/?p=amarok.git&a=commit&h=71e55e7bdd9b31642d09145a5801777c0e120454

However, at the moment https://github.com/Ultimaker/Cura/issues/402 is more important...

daid, 2016-01-16:
For windows I have this code in the old Cura:
https://github.com/daid/Cura/blob/SteamEngine/Cura/gui/printWindow.py#L12

Ghostkeeper, 2016-01-16:
If we add a type of job that is "critical" then we can generalise this to also happen while doing other critical jobs, such as writing to devices and other things that shouldn't be interrupted that we'll do in the future.

ianpaschal, 2018-02-27:
Hasn't been touched in almost 2 years.

closed 2018-02-27 by ianpaschal