Ultimaker/Cura #599: Crash on startup (Mac OS 10.9.5)

motenay, 2016-01-18:
I'm not sure on what to do...

Here is all the data I have.

Version: 15.06.03
Platform: Darwin-13.4.0-x86_64-i386-64bit
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "/Applications/Cura.app/Contents/Resources/cura_app.py", line 10, in <module>
  File "cura/CuraApplication.pyc", line 176, in run
  File "UM/Application.pyc", line 246, in setActiveMachine
  File "UM/Signal.pyc", line 96, in emit
  File "cura/CuraApplication.pyc", line 503, in _onActiveMachineChanged
  File "cura/BuildVolume.pyc", line 144, in rebuild
TypeError: can't multiply sequence by non-int of type 'float'

Ghostkeeper, 2016-01-19:
You can probably work around this by using a more recent build. The most recent OSX build that Ultimaker published was from back when the public beta was going on: http://software.ultimaker.com/Cura_closed_beta/15.10/Cura-1.99.0-dev.20151117-Darwin.dmg

It looks like you've imported some other machine that has a wrong setting.
If that's not the case, I'll venture to guess that it was a problem with the version you're using and has been fixed in the meanwhile :smiley: Check out that new version and see if it persists.

motenay, 2016-01-19:
I did not load any other machine, actually it was a clean installation...
Thanks a lot!
This seems to work (even if there are some minor problems, such has I had to change the resolution of the monitor to get to the "confirm" button of the screen to add a machine and all the properties... :D). I did not know there were more updated Beta to have fun with.

Ghostkeeper, 2016-01-19:
Then I'll assume that it has been fixed in the meanwhile.

The size of some of the windows has been fixed in more recent versions too (more recent than the one I just pointed you to). We have two such issues left unfixed: The profile window is too small and the drop-down at the save-to-file button goes off screen. But those are already addressed in other issues here.

closed 2016-01-19 by Ghostkeeper