Ultimaker/Cura #600: "'NoneType' object has no attribute 'getGCode'" when slicing fails

Feuermurmel, 2016-01-18:
Cura 15.04.4 crashes with an `AttributeError` when trying to slice an object that's too large:

```
$ /.../Cura.app//Contents/MacOS/Cura --slice --ini /.../stuff/profile.ini --output /.../src/foo.gcode /.../src/foo.stl
load preferences from /.../Library/Application Support/Cura/15.04/preferences.ini
Failed to listen on port: 49674
Listening for engine communications on 49675
Traceback (most recent call last):
  File "/.../Cura.app/Contents/Resources/__boot__.py", line 316, in <module>
    _run()
  File "/.../Cura.app/Contents/Resources/__boot__.py", line 311, in _run
    exec(compile(source, path, 'exec'), globals(), globals())
  File "/.../Cura.app/Contents/Resources/cura.py", line 89, in <module>
    main()
  File "/.../Cura.app/Contents/Resources/cura.py", line 75, in main
    gcode = engine.getResult().getGCode()
AttributeError: 'NoneType' object has no attribute 'getGCode'
2016-01-18 20:56:34.722 Cura[30505:507] Cura Error
```

Ghostkeeper, 2016-01-19:
For issues of version 15.04 and below I'd like to refer you to https://github.com/daid/Cura/issues

The repository was moved when it was rewritten from scratch after 15.04. We still support 15.04, but we'd like to keep the issues separate.

closed 2016-01-19 by Ghostkeeper

Feuermurmel, 2016-01-19:
Oh snap. I see that this issue may have already been reported: https://github.com/daid/Cura/issues/1105

commit 92679818 (2017-07-27): Added min width & height

FIXES #600