Ultimaker/Cura #601: Cura can't find file passed on command line

Feuermurmel, 2016-01-18:
Cura 15.04.4 crashes with an `IOError` on an existing file:

```
michi ~/.../Designs/Material_Test_Shapes$ /users/michi/Applications/Cura.app//Contents/MacOS/Cura --slice --ini stuff/profile.ini --output src/foo.gcode src/foo.stl
load preferences from /Users/michi/Library/Application Support/Cura/15.04.4/preferences.ini
Listening for engine communications on 49674
Traceback (most recent call last):
  File "/users/michi/Applications/Cura.app/Contents/Resources/__boot__.py", line 316, in <module>
    _run()
  File "/users/michi/Applications/Cura.app/Contents/Resources/__boot__.py", line 311, in _run
    exec(compile(source, path, 'exec'), globals(), globals())
  File "/users/michi/Applications/Cura.app/Contents/Resources/cura.py", line 89, in <module>
    main()
  File "/users/michi/Applications/Cura.app/Contents/Resources/cura.py", line 67, in main
    for m in meshLoader.loadMeshes(args[0]):
  File "Cura/util/meshLoader.pyo", line 32, in loadMeshes
  File "Cura/util/meshLoaders/stl.pyo", line 54, in loadScene
IOError: [Errno 2] No such file or directory: 'src/foo.stl'
2016-01-18 21:07:30.245 Cura[31285:507] Cura Error
michi ~/.../Designs/Material_Test_Shapes$ ls src/foo.stl
src/foo.stl
```

The file at `src/foo.stl` exists and is readable.

fieldOfView, 2016-01-18:
Thanks for your reports. This is the repository for the "new" Cura. Issues for the old Cura (up to 15.04.x) should be logged on https://github.com/daid/cura

daid, 2016-01-19:
Old Cura needs absolute paths, not relative.

closed 2016-01-19 by daid

Feuermurmel, 2016-01-19:
I've created a new issue: https://github.com/daid/Cura/issues/1399