Ultimaker/Cura #602: Cura prints stack trace even when slicing is successful

Feuermurmel, 2016-01-18:
Cura 15.04.4 displays a stack trace even though slicing was successful.

```
$ /.../Cura.app//Contents/MacOS/Cura --slice --ini /.../stuff/profile.ini --output /.../src/foo.gcode /.../src/foo.stl
load preferences from /.../Library/Application Support/Cura/15.04.4/preferences.ini
Failed to listen on port: 49674
Listening for engine communications on 49675
Exception in thread Thread-2:
Traceback (most recent call last):
  File "threading.pyo", line 551, in __bootstrap_inner
  File "threading.pyo", line 504, in run
  File "Cura/util/sliceEngine.pyo", line 416, in _runEngine
  File "Cura/util/sliceEngine.pyo", line 93, in getPrintTime
NameError: global name '_' is not defined

GCode file saved : /.../src/foo.gcode
```

nallath, 2016-01-19:
This is an issue for the old version of Cura, which can be found on https://github.com/daid/cura

closed 2016-01-19 by nallath

Feuermurmel, 2016-01-19:
Thanks. I've opened an issue in the correct project: https://github.com/daid/Cura/issues/1400